svenhb/GRBL-Plotter #156: Backward streaming

BehiTaj, 2020-10-20:
**Adding Backward in streaming**
Sometimes in between file streaming(run gcode file) we need to hold streaming and then backward the file.
Is there any feature for do that???
thanks :)

svenhb, 2020-10-22:
a) Do you need to go backwards line by line? 
b) Or do you need to hit a previous XY-point given by one code-line?
For b) you can use 'pause' , better lift Z axis now, then click into the 2D-view to set the marker, then rigth click and select 'Move to marked position G0'
[image]

BehiTaj, 2020-10-22:
Backward line by line. 
Actually i want to do following steps:
1 -run/play a file
2- feed hold between file
3- backward line by line from holded position to 
the start of file

svenhb, 2020-12-10:
With normal grbl it is not possible I think:
2) feed-hold just pause the execution, but doesn't clear the buffer, you need to reset the controller, otherwise it would continue the program: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions

To go backwards requires to replace G2/G3 commands which would change your current GCode.