svenhb/GRBL-Plotter #158: The software source code of version 1.5.2.2 cannot be compiled normally

lianzaozi, 2020-11-03:
When using Visual studio software to compile the source code of the plotter of version 1.5.2.2, it cannot be compiled normally, and visual studio will report an error.

svenhb, 2020-11-03:
Can you show the error here? Probably an outdated file.

lianzaozi, 2020-11-03:
> 您可以在这里显示错误吗？可能是过时的文件。

The error message is as follows:

CS1061'"Settings" does not contain the definition of "sizeSerForm1", and the accessible extension method "sizeSerForm1" that can accept the first "Settings" type parameter cannot be found (are you missing a using directive or assembly reference?) GRBL-Plotter D :\GRBL-Plotter-V1.5.2.2 20201025\GRBL-Plotter-master\GRBL-Plotter\MachineControl\ControlSerialForm.cs 166 Active
[image]

CS1061'"Settings" does not contain the definition of "sizeSerForm1", and the accessible extension method "sizeSerForm1" that can accept the first "Settings" type parameter cannot be found (are you missing a using directive or assembly reference?) GRBL-Plotter D :\GRBL-Plotter-V1.5.2.2 20201025\GRBL-Plotter-master\GRBL-Plotter\MachineControl\ControlSerialForm.cs 387 Active
[image]

svenhb, 2020-11-03:
I forgot to upload the settings files, check here: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/Properties

lianzaozi, 2020-11-04:
It will now compile normally

closed 2020-11-21 by svenhb