svenhb/GRBL-Plotter #161: Freeze within setup menu

amoineau, 2020-11-12:
**Describe the bug**
I downloaded the latest version of GRBL-plotter (1.5.2.2) and I started changing things in the setup menu (Working area, DXF to Gcode settings, tool table, etc..). Every time I click on a tick box or change a text field the application freezes. It can vary from a second to a full minute.

I wanted to know if it was a known behavior or if the problem was specific to my setup and if you had any idea on how to solve this ?

_I know I have been requesting a lot of help from you recently and I'm really grateful for your responsiveness and your hard work._

svenhb, 2020-11-12:
Did you run it from VisualStudio?
It's the same for me, I don't remember which debug option must be switched off - I will figure out.

amoineau, 2020-11-12:
Yes I realised after posting that it might be the source of the problem. I started trying from the bin/Debug and Release .exe and indeed it seems to work fine there !

amoineau, 2020-11-12:
But if I change the setup there it will not be persistent ?

svenhb, 2020-11-12:
The settings should be saved in C:\User\myName\AppData\Local\GRBL_Plotter\subfolder

amoineau, 2020-11-18:
It seems that saving a usecase when I change things and loading it when I want to apply the changes is a viable option. Does it seems right to you ?
It's just a bit tedious to move the usecase around and make sure that it is up to date. But that way I do the the setup through the .exe version and I can still use VS to edit things.

svenhb, 2020-11-18:
I still need to check the debug-option, which needs to be swichted off, to speed up the setup form.  
You could save your INI-files on the desktop and just drag and drop them on GRBL-Plotter, as shown here: https://youtu.be/2Cc7sjwkxK8

amoineau, 2020-11-18:
About that, I was wondering: does machine config includes everything in the current use case ? Because I've been saving and importing both every time but by reading them it feels like I could just use the global config file.

svenhb, 2020-11-18:
Saving the "machine config" includes almost all settings (including custom buttons).  
"Use case" should save only import options. You could extend this files (via text editor) with settings from the machine config. 
Reading is the same for both: if set 
```
[Info]
Set Defaults=True
```
All import settings will be set to default. Check file \MachineControl\IniFile.cs

svenhb, 2020-11-21:
This may help (https://stackoverflow.com/questions/12567984/visual-studio-debugging-loading-very-slow):
- Go to Tools -> Options -> Debugging -> General  
- CHECK the checkmark next to "Enable Just My Code".

amoineau, 2020-11-24:
Ok I'll try that !

amoineau, 2020-12-03:
It works but now I get these kind of messages every time something happens : 
[image]

amoineau, 2020-12-03:
But it's not really a problem to be honest, I compile then if I want to debug (more like use breakpoints to understand how things work) I run and if I want to use it for real or adjust setup, I open with the .exe .

svenhb, 2020-12-03:
You may try to fix it with this code (I can't try it out until tonight):  
Find function "private void updateProgressBar" in file "MainFormStreaming.cs"  
Replace this two lines: 
```
pbFile.Value = cPrgs;
pbBuffer.Value = bPrgs;
```

by:  
```
        //    pbFile.Value = cPrgs;
            if (this.pbFile.InvokeRequired)
            {   this.pbFile.BeginInvoke((MethodInvoker)delegate () { this.pbFile.Value = cPrgs; }); }
            else
            {   this.pbFile.Value = cPrgs;   }

        //    pbBuffer.Value = bPrgs;
            if (this.pbBuffer.InvokeRequired)
            {   this.pbBuffer.BeginInvoke((MethodInvoker)delegate () { this.pbBuffer.Value = bPrgs; }); }
            else
            {   this.pbBuffer.Value = bPrgs;   }

```

amoineau, 2020-12-03:
Ok, though I gave you this one has an example but I get many different ones ^^'.
I can usually skip them by doing "continue" but when it happens in a loop....
So I decided to uncheck "just my code" and work like that.

closed 2020-12-08 by amoineau