svenhb/GRBL-Plotter #162: Link a command/script to a specific tool

amoineau, 2020-11-13:
**Context**
I want to be able to send the command "M9" after "T1 M6" and "M7" after "T2 M6" because I overrided this command to control a relay to divert the PWM between my 2 tools.

My first thought was to use the "Tool Change" setup but I couldn't find a way to make the script conditional (depending on the new tool).

I then noticed the Gcode column in the tool table which seemed to fit my need.

**Problem**
The tool tip says "code to apply after tool change command"

[image]

So I added my commands but nothing happens after Tx M6 when I import the DXF. I checked and the .csv is correctly saved. However I might just have misunderstood what that column was about.

**Screenshots**
Here I import a DXF with 2 squares, one on a layer called "Laser" and one on a layer called "Encre". The tool matching features works great by the way (thank you for that !).

[image]

svenhb, 2020-11-13:
I checked the code: the gcode from tool table will only be added if "Set XY-Feed from tool table" is enabled.
[image]

amoineau, 2020-11-13:
OK I'll try !

amoineau, 2020-11-13:
Working great thank you again !
Can I put several commands in the gcode column of the Tool Table ?

amoineau, 2020-11-13:
Huh, it's a different problem but the command M7 is not recognized and return an error. M8 and M9 work fine.
Do you want me to open a new issue or can we attend this here ?

[image]

svenhb, 2020-11-13:
Yes, line seperation by ';' as for custom buttons

amoineau, 2020-11-13:
Ok thanks.
The M7 problem seems to come from GRBL itself, but it worked not long ago... weird but maybe not your problem.

svenhb, 2020-11-13:
This is an error from grbl... See description:
[image]

amoineau, 2020-11-13:
Ok I have a grbl version mismatch between my debug board and the machine board...
But initial problem solved !

amoineau, 2020-11-13:
Graphic import > Path import > Grouping and Tools > Tool Table
"Set XY Feed..." needs to be ticked.

closed 2020-11-13 by amoineau