svenhb/GRBL-Plotter #163: Questions about Tool Change

amoineau, 2020-11-18:
Hello, I have two tools attached on my machine and I use M7-M9 to control a relay to send the Spindle PWM to either one.
I match the DXF layers with the tools so that grbl plotter will automatically detect tool changes and use the right tool plus send the M7-M9 command. https://github.com/svenhb/GRBL-Plotter/issues/162

Here is an example with two squares supposed to be drawn with one tool each:

[image]

1) Is there a way to remove the pause before and after each tool changes, I couldn't find the setting that was causing it. I removed every scripts in the tool changes menu so I don't think it's coming from there. 

2) Obviously the tools are not at the same position so i tried to change the X and Y value in the tool table but it didn't change anything. Do I have to manually apply the offset through a script maybe ? Or should I use a different referential, like G54 for tool 1 and G55 for tool 2, and set the offset in G55 ? What is the best practice according to you ?
3) I don't understand how this works :

[image]

Compared to changing the values in the tool table : 

[image]

When I click on "Move to position.." it wants to move tool 1 to 50;120 and tool 2 to 50+table_offset;120+table_offset. So it seems to apply a base offset plus the value in the table but in practice none of them seem to have an offset.

Thanks again for your help and your work.

svenhb, 2020-11-18:
1) There is no setting to remove the pause - up to now. You can remove it by your own in file \GCodeCreation\gcodeRelated.cs lines 1021, 1042  
2) Using G54, G55 is recommended if you use laser and pen, as I showed here: https://grbl-plotter.de/index.php?id=laser-pen&setlang=en  
Real tool positions are needed if you use scripts to take a pen, as showed here: https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en  
3) It is just a "global" offset for the positions inside the tool table 

Hope this helps

amoineau, 2020-11-18:
It helps a lot, thank you ! 
Is the global offset activated by default ? Was I working with the offset without realising ?

svenhb, 2020-11-18:
Yes, the global offset is alway activated. 
**Note: all these positons are _machine_ positons, not work positions**

amoineau, 2020-11-18:
I'm not sure I get the difference...

svenhb, 2020-11-18:
If you save a "use case" here:
[image]

It will generate an INI-file but with reduced content - compared to "Export machine settings"
[image]

amoineau, 2020-11-18:
No sorry I meant between machine position and work position !
I understood the part about usecase and machine settings ^^

svenhb, 2020-11-18:
If you use Homing, the machine position will be set and is different to the work position.
It is not possible to change the machine position (except you move XYZ).
Then tool positions are absolute - for ever, no dependence on G54 etc.

amoineau, 2020-11-18:
Ok I think I understand. I used the G54,55 solution and it messed with my head at first but I got it now and it works great.

amoineau, 2020-11-19:
Last question then I close the issue (slightly off topic) :
One of my tool is a micro dispensing valve to print with ink, I'd like to be able to replace small circles by a dot because under 5mm it just makes a big bubble of ink if i try to make a circle. Is there a way to do that ?

svenhb, 2020-11-19:
I made a conversion circle-radius to Z-value - but also not really tested. The idea was to translate stipple -pictures https://www.evilmadscientist.com/2012/stipplegen2/:
[image]

amoineau, 2020-11-24:
Unfortunately it doesn't change anything for me. 
Here is a part of the DXF file :

[image]

In white is the laser layer and in red the ink one. The little circles have a 2mm diameter so I'd like to simply drop a droplet at the center. It would be much faster and more precise.

If I were to edit the source code to add a filter to transform these circles, where would the proper place be in the code ?
I figured it might be best to filter when streaming and not importing, that way the drawing in the viewer would keep its appearance. 

Finally, by any chance, do you know what Gcode would allow me to simply make a pulse at a given position (no circle, no line, just a dot) ? 

Thank you for your help, I would never have gone that far without it !

svenhb, 2020-11-24:
The GCode to make a dot is just "Pen down" / "Pen up".
I predefined buttons for z-down, z-up - just check subfolder ..data/buttons.

If you have the DXF source: In LibreCAD you can apply Points instead of circles - which results to a Z-down/-up process:
[image]

amoineau, 2020-11-24:
I changed the circles to points and the movements seem right but the Spindle is not triggered. I get the following result :

[image]

When I was using circles it happened sometimes but now it happens on every point. 
I have "Remove very short move" disabled and "Distance to be assumed as equal" to minimum.

Another weird glitch is that the feed rate is not provided anymore... The same file with circles gave this :

[image]

It's not a big problem, I can manually fix it, but maybe it can help understand what's happening.

svenhb, 2020-11-24:
There are two different feedrates possible: for XY and for Z.  
You don't use Z, so there is no need to switch between XY- and Z-feedrate, it's always the same feedrate, which should be set at the beginning of the GCode. 
What do you mean with "the spindle is not triggered"? If you expect spindle on/off on pen-down/-up, you need to tick that option:
[image]

amoineau, 2020-11-24:
I mostly never touched Z feedrate, I only worked with XY. I use XY feedrate frome tool table as well as spindle.
Yes I expect a Spindle On/Off and I have this option ticked already :(

svenhb, 2020-11-24:
Ok, I need to check the code...
Workarround: add M3, M5 to the individual commands and enable them...

amoineau, 2020-11-24:
Maybe something is wrong in my setup, I'm using the following :

[image]

[image]

[image]

[image]

[image]

svenhb, 2020-11-24:
This two options "bites" each other, if you want M3 / M5 the "Lasermode" must not be ticked...
[image]

If you don't do any tool change, I think this two options are not needed:
[image]
[image]

amoineau, 2020-11-24:
(I do use tool change "virtually", it allows me to change the setting (Feed, spindle, position, M3/M4, PWM output, etc...) between the laser and the ink dispenser.  If I uncheck "Perform tool change on Tx M06" I get an error from GRBL because the Tx M06 is not removed and is not recognized by GRBL.)

I unticked "Lasermode" and I get the following result :

[image]

At first it ddin't work but then I realised that it couldn't work with GRBL laser mode ($32) on because I was static. So I tried without and it works. But I have a few issues with this solution. 
The main one is that I do need the Laser Mode with M4 to have acceptable result with the laser. Additionally, it adds unecessary delays before each figure when doing the laser layer.

svenhb, 2020-11-24:
How about enabling the Z axis (without having one connected) to create movement - perhaps Lasermode M4 then works.
To switch on/off the laser with "Lasermode" ticked on, try to add the "Individual commands" with M4 / M5...

svenhb, 2020-11-24:
>I do use tool change "virtually", it allows me to change the setting (Feed, spindle, position, M3/M4, PWM output, etc...) 

It should already work just with this setting:
[image]

amoineau, 2020-11-24:
> > I do use tool change "virtually", it allows me to change the setting (Feed, spindle, position, M3/M4, PWM output, etc...)
> 
> It should already work just with this setting:
> [image]

It does work but it prints an error :

[image]

(I asked the Hold)

But it seems consistent with your description :

[image]

(Last line)

amoineau, 2020-11-24:
> How about enabling the Z axis (without having one connected) to create movement - perhaps Lasermode M4 then works.
> To switch on/off the laser with "Lasermode" ticked on, try to add the "Individual commands" with M4 / M5...

Good idea, I'll try that tomorrow.

amoineau, 2020-11-25:
Unfortunately Z moves don't seem to trigger the spindle in laser mode, which is logical when you think about it, since the purpose is to not keep the laser on over a single spot...

svenhb, 2020-11-25:
Then try to use "Individual commands" and add a tiny relative move back and forth. Like G91 G1 X0.01 and G91 G1 -X0.01 for up and down.

amoineau, 2020-11-25:
Yep that's exactly what I was doing and it seems to be ok

amoineau, 2020-11-25:
But I was afraid that switching to incremental movements could be a problem ?

svenhb, 2020-11-25:
Ok, then write "G91 G1 X0.01;G90" to switch back to G90 after the move. ";" should start a new command line.

amoineau, 2020-11-25:
I tried that and multiple other combinations :/
And I just realised that it won't be a solution anyway because I can't have these commands when using the second tool (laser). I'm really sorry to be so annoying with my weird requirements... Thank you for your patience !

What I think would be the ideal solution would be to choose how a DXF Point is treated, can you point me toward where its handled in the code ? Because I simply want to replace points by a small line, almost like it's already doing by default without using pen up/down, I just need to make it slightly bigger I think (and only one way).

svenhb, 2020-11-25:
Probably you want to change the dot on an early point of import, so I would suggest GCodeCreation\ImportFromFile\GCodeFromDXF.cs Line 338.

amoineau, 2020-11-25:
Ok thanks !

amoineau, 2020-11-25:
I replaced :
`GCodeDotOnly(position, "Start Point");`
with : 
`DXFStartPath(position, "Start Line");
DXFMoveTo(pos2, "");`
To make small dashes instead of dots, works great on simulation, i'll try on the machine during the next batch of tests.

amoineau, 2020-11-25:
Do you think it will mess up something else ? Like interfering with G0 commands or other commands that might use dots?

svenhb, 2020-11-25:
from where you get pos2? -> `DXFMoveTo(pos2, ""); ` 
It just interfers with the import of DXF content. I use "DXF-Points" for drills an would not change the code as you do.  
Finally I don't know what you want to get as GCode, when "inserting" DXF-Point.

amoineau, 2020-11-25:
> `DXFStartPath(position, "Start Line"); DXFMoveTo(pos2, "");`

Here I copy-pasted the line from Line import just to explain. I used position for both start and end but I shifted the Y for both (+/-) to make a small dash "across" the dot.

svenhb, 2020-12-10:
I think this question was answered.
About Pen-up / -down algorythm, I made this flow charts:
https://grbl-plotter.de/data/uploads/pic/flow_charts/Pen_down.png
https://grbl-plotter.de/data/uploads/pic/flow_charts/Pen_up.png

closed 2020-12-10 by svenhb

amoineau, 2020-12-11:
yes indeed, thanks