svenhb/GRBL-Plotter #166: It freezes when loading heavy dxf or svg files

laoshipoto, 2020-11-29:
Well, hello 🖐️, I've never posted in github. But anyway 😅
I have heavy dxf and svg files, like too many lines to plot.
When I load these files the software freezes. And I have to reopen it.
Is there any solution to make It open these kind of heavy files?
Some other software like "engraver master" or "bachin draw "actually do struggle for like 10 secs but it will open eventually.
Thank for your great job, thank you for your help, thank for your time
Vielen Dank🙏😊

svenhb, 2020-11-29:
Hello,
1) try to import with all import options switched off - see blow, also without path modifications, clipping, grouping.  
2) try an older version, before 1.5.0.0. In this version I made a big change in handling of the imported graphic objects.  

I hope this helps.  
Perhaps you can upload such a big file here? 

Import options awitched off: 
[image]

laoshipoto, 2020-11-29:
[jojo_HD.zip](https://github.com/svenhb/GRBL-Plotter/files/5612718/jojo_HD.zip)
hello thanks for replying :)
turning off those option did nothing. but here and exemple of a file.
Thanks again🙏

svenhb, 2020-11-29:
Yes, it won't load as DXF, but I opened your file in LibreCAD and exported it as SVG. This could be loaded in GRBL-Plotter:
[image]

laoshipoto, 2020-12-11:
Hello, the software dis open that once but keeps freezing everytime. Maybe adding more "memory". I don't know really. I just wanted to give my feedback.
Thank you for your hard work and time 🙏

svenhb, 2020-12-11:
Thanks for the feedback, I could solve the problem by showing the progress of loading with option to abort - not released yet.
Very very time consuming is the option "Sort objects for shortest connection and merge".
Also to fill the text-editor with created GCode takes some seconds ...

svenhb, 2020-12-18:
Please check new version: https://github.com/svenhb/GRBL-Plotter/releases

closed 2021-01-17 by svenhb