svenhb/GRBL-Plotter #170: The trace shows that some strange problems have occurred

lianzaozi, 2020-12-04:
When processing the letter "B" and letter "L", there are some problems with the track displayed by the software.

[GCode.zip](https://github.com/svenhb/GRBL-Plotter/files/5642393/GCode.zip)

[image]

[image]

svenhb, 2020-12-04:
What kind of computer and operating system are you using? Maybe it is out of date?
Do you get errors inside the COM window?

lianzaozi, 2020-12-04:
> 您使用哪种计算机和操作系统？也许已经过时了？
> 您是否在COM窗口中收到错误？

OS: windows10 X64
computer: notebook

lianzaozi, 2020-12-05:
Does the reason for this problem have anything to do with the processing speed in GCode?

svenhb, 2020-12-05:
Yes, as faster the movement as more complicate to display the processed path, because tool-position and gcode-line-nr will be updated to main GUI with 2 Hz to max 10 Hz:
[image]

lianzaozi, 2020-12-07:
This is the configuration information of my grbl control board that I used the plotter software to query. The information shows that the maximum speed allowed is 500. Now I set the processing speed in GCode to 100, but there are still strange things in the process of drawing the processing track line.

[image]

> $$
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=25          (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=0           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=0          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=0          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=0          (Enables homing cycle. Requires limit switches on all axes.)
< $23=0          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=25.000     (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=500.000    (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=1.000      (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=1000       (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=250.000   (X-axis travel resolution in steps per millimeter.)
< $101=250.000   (Y-axis travel resolution in steps per millimeter.)
< $102=250.000   (Z-axis travel resolution in steps per millimeter.)
< $110=500.000   (X-axis maximum rate. Used as G0 rapid rate.)
< $111=500.000   (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=500.000   (Z-axis maximum rate. Used as G0 rapid rate.)
< $120=10.000    (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=10.000    (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=10.000    (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=200.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=200.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=200.000   (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok

lianzaozi, 2020-12-07:
Now my speed reduction problem still occurs, I hope you can test it with the GCode I uploaded.

svenhb, 2020-12-18:
If I run the program inside Visual Studio in Debug-Mode I get the same problem, but not in Release-Mode or without Visual-Studio.

closed 2021-05-02 by svenhb

svenhb, 2021-10-09:
Improoved behavior of progressed path display: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_6_2_1.exe