svenhb/GRBL-Plotter #172: The source code for Version 1.5.3.0 does not compile.

lianzaozi, 2020-12-19:
Thank you for releasing the new Version. I just downloaded the source code of Version 1.5.3.0, but it cannot be compiled in VS.

lianzaozi, 2020-12-19:
V1.5.3.0 source code click on the vs menu bar "generate "->" rebuild solution ",vs will report an error.

svenhb, 2020-12-19:
>Version v1.5.3.0 of the software does not support opening by dragging and dropping dxf files.  

For me it's working - strange, also downloading https://github.com/svenhb/GRBL-Plotter/archive/v1.5.3.0.zip works.

lianzaozi, 2020-12-19:
Visualstudio will still report an error when regenerating the solution. I used visualstudio 2019 as my IDE.

But press the F5 shortcut to run the source code, which does not report errors

svenhb, 2020-12-19:
I use Visual Studio 2015...  
>Visualstudio will still report an error   

What kind of error?

lianzaozi, 2020-12-19:
[image]
[Error information.zip](https://github.com/svenhb/GRBL-Plotter/files/5719478/Error.information.zip)

svenhb, 2020-12-19:
Sorry, no idea, what's wrong.  
Older version 1.5.2.4 works?

lianzaozi, 2020-12-19:
The source code of version 1.5.2.4 will also report an error when regenerating the solution. I guess this error has nothing to do with the source code of the program. I am looking for the cause of the error.

closed 2021-01-17 by svenhb