svenhb/GRBL-Plotter #173: grbl mega 5 axis compiled hex with limit

78Matteo78, 2020-12-24:
Hello
i can't get hard limits to work with grbl mega 2560
Could it be that the compiled version of 2019 doesn't work?
it is possible to have the latest version which should have active limits.
the pins of the limits are D2, D3, D14, D15, D18, D19.

svenhb, 2020-12-25:
I never tried this option with my mega 2560. 
You may follow this text: https://github.com/fra589/grbl-Mega-5X/releases/tag/v1.1l.20190605
- Download the latest files here https://github.com/fra589/grbl-Mega-5X
- And follow the instructions here, how to compile the source inside the Arduino-IDE: https://github.com/gnea/grbl/wiki/Compiling-Grbl

futurau, 2020-12-25:
grbl-Mega-5X is working great. I am using it on my cnc. I am not using max limits. Only min limits for homing (NC contact switches). []https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout(url)

78Matteo78, 2020-12-25:
Merry Christmas
I solved it by making a change on the cpu_map.h enable hard limit library, as explained in the firmware site, now they work.
thanks
greetings matteo

svenhb, 2020-12-25:
👍👍👍
If you like, I can upload your Hex File here

closed 2020-12-25 by 78Matteo78