svenhb/GRBL-Plotter #175: Incorrect apply to gcode approximated height map

alligator77, 2020-12-29:
Incorrect apply to gcode approximated height map

Source gcode
[main34-source.txt](https://github.com/svenhb/GRBL-Plotter/files/5752423/main34-source.txt)
Map
[map.txt](https://github.com/svenhb/GRBL-Plotter/files/5752415/map.txt)

if try to apply this map to main34, then generated bad code:
[main34_bad.txt](https://github.com/svenhb/GRBL-Plotter/files/5752443/main34_bad.txt)

Without height map:
[image]
[image]

After apply height map:
[image]
[image]

svenhb, 2020-12-30:
Your source code looks a bit wired to me:
If I just go through the code in GRBL-Plotter and watch the position marker:
why this jumps in code lines 974 (N2032), 1061 (N2206), 1145 (N2374), 1226 (N2536), 1290 (N2664) for the first element?

It seems the gcode was stiched together - real line numbers and N-numbers are not matching.

The real problem for my algorithm is probably the already given Z in XY move statring in line 253 (N590)

Please try with a more clean gcode -> XY and Z moves seperated.

alligator77, 2020-12-30:
> 
> why this jumps in code lines 974 (N2032), 1061 (N2206), 1145 (N2374), 1226 (N2536), 1290 (N2664) for the first element?
> 
Yep, this code contains 3 operations for one end mill, but this is normally. And transormed for cutting 6 pieces.

> The real problem for my algorithm is probably the already given Z in XY move starting at line 253 (N590)
Probably yes. this moving is spiral cutting, with smooth Z adjustment.

> Please try with a more clean gcode -> XY and Z moves seperated.
This source code is ok, and testing with Candle. But You software is better for me, and And I hope to completely switch to it.

I am also a .Net developer, but only the day before yesterday I found your software for myself, and have not had time to dig into it yet. :)

svenhb, 2020-12-30:
BTW: grbl for Arduino Uno doesn't support G43 H2... Up to now "H2" will be removed by GRBL-Plotter

alligator77, 2020-12-30:
> 
> 
> BTW: grbl for Arduino Uno doesn't support G43 H2... Up to now "H2" will be removed by GRBL-Plotter

Yep, some commands grbl not supported, these commands are ignored by grbl and do not affect the result

svenhb, 2020-12-30:
Please check new release: https://github.com/svenhb/GRBL-Plotter/releases

alligator77, 2020-12-30:
Tnx very mutch! Now it is fine!

closed 2021-01-17 by svenhb