svenhb/GRBL-Plotter #176: Rotation gcode in Camera window is failed

alligator77, 2021-01-19:
If I try to rotate to any angle ("Apply Angle To GCode" button in Camera Window), all path is disappears. 
But if I rotate GCode from Top menu, that works fine.

**Reproduce**
Video
https://user-images.githubusercontent.com/6526798/105091477-ef45b900-5ab0-11eb-94e0-73210ed33f32.mp4

svenhb, 2021-01-19:
I haven't tried the camera functionality in a long time - I'll check it out and repair.
Thanks for the feedback

alligator77, 2021-01-22:
Pull requested

svenhb, 2021-01-22:
Thanks, there are some more bugs to fix: e.g. importing the drill file doesn't work. 
I need to check the described work flow: https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling

svenhb, 2021-01-22:
Should be fixed now: https://github.com/svenhb/GRBL-Plotter/releases

alligator77, 2021-01-22:
Now works fine! Tnx

svenhb, 2021-03-08:
Thanks for testing

closed 2021-03-08 by svenhb