svenhb/GRBL-Plotter #177: Rotation Gcode arround camera cross center

alligator77, 2021-01-19:
Can add feature to rotate GCode arround camera cross center without activate "Auto alignment"? Tnx

[image]

svenhb, 2021-01-19:
Yes should be possible

alligator77, 2021-01-20:
> 
> 
> Yes should be possible

Tnx.

One more question. Are you planning to implement 3D previews?

svenhb, 2021-01-20:
Sorry, no 3D view in planning.

svenhb, 2021-01-22:
Try https://github.com/svenhb/GRBL-Plotter/releases

alligator77, 2021-01-22:
Sorry, a little bit wrong, has need to turn not from 0: 0, but from the center of the yellow cross, and it can be anywhere.

[image]

svenhb, 2021-01-23:
I would follow this description: https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling
It should work like this...
0) reset offet (right click in camera view G92.1)
1) Switch to 'Cam coord.'
2) select marker = center of rotation in 2D-view in main GUI
3) move camera view to same position on real object
4) In camera view, right click on  select 'Set actual pos. to marker pos.'
5) in 2D view select next marker as reference for rotation and scale
6) move camera view to same position on real object
7) In camera view, right click on  select 'Compensate angle and scale'

alligator77, 2021-01-24:
Ok, I check this out

closed 2021-05-02 by svenhb

engrjawadulhassan, 2022-01-11:
Hello,
I hope every one doing great.

I am having issues , with orientation of G code for drilling an etched PCB. 
I was using the latest version of GRBL plotter for this task, but when i reached the tutorial , the tutorial was for a previous version the software. Than , i downloaded the previous version 1.5.2 , it give me access to same camera window with tool cord and cam cord selection, I have followed the whole tutorial and steps as described.
https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling

But still i am having an error while performing the drill JOB , drill is slightly out from the drill point on pad ob both axis (X and Y), and this error is random for each point. 

One more thing , that i noticed is, In new version this tutorial doesn't help at all , due to changes in software menus and terms , that were in previous version, so in my opinion tutorials/guides also needed an update regarding the changes in software.

Need help regarding this issue, Please guide to any solution.

Thanks

svenhb, 2022-01-11:
In principle you  
- move to  first reference point and right click "Add reference point", then move to 
- move to  2nd ref-point and right click "Add reference point" 
- finally press "Start automatic fiducial correction"   
[image]

engrjawadulhassan, 2022-01-12:
Hello,

What I am doing is:
1.Connect GRBL
2.Load GCODE
3.Open Camera Window
4.Followed your described steps as above

I did the same as you told , but still i am facing the same issue regarding accuracy of new drill point position.
So is there any other step related to this procedure , that i am missing. 

Thanks

svenhb, 2022-01-12:
I forgot to mentioned: you need the same amount of fiducial coordinates (from 2D-view) and according reference points (from camera view)