svenhb/GRBL-Plotter #180: Something is freezing the UI and main thread

alligator77, 2021-02-01:
If send gcode commands via console, it is all fine. But if execute codes from UI (see video), something doing wrong, freeze UI for all app of few seconds for each one command. 

If I run the file with multiple commands, then the UI hangs for every line.

I don't know how this reproduce, but this happens in Control coordinate systems window or run predefined Gcode files in "Custom Buttons" section.
I repeat, if send codes in console, all works fine.

First I checked all the USB cables, I thought it was a problem with them, but other senders work fine.

https://user-images.githubusercontent.com/6526798/106493533-4e042d00-64ca-11eb-894c-77b344083218.mp4

svenhb, 2021-02-01:
Thanks for the detailed description.
As long as the "Control Coordinates System" is open, also main GUI "Zero X/Y/Z" will cause freeze

alligator77, 2021-02-01:
> 
> 
> Thanks for the detailed description.
> As long as the "Control Coordinates System" is open, also main GUI "Zero X/Y/Z" will cause freeze

Yep, This freeze too

alligator77, 2021-02-01:
> 
> 
> Thanks for the detailed description.
> As long as the "Control Coordinates System" is open, also main GUI "Zero X/Y/Z" will cause freeze

And without Control Coordinates System window same issue if call button

https://user-images.githubusercontent.com/6526798/106507263-d8a15800-64db-11eb-9e48-571e4d676010.mp4

Button has link to file with this commands:
`G54
G10 L20 P0 X0 Y0
G59
G10 L20 P0 X0 Y0

G54
G90 G0 X-53.558 Y23.605 F80
G10 L20 P0 X0 Y0

G59
G0G90 X0 Y0`

svenhb, 2021-02-01:
I made a quick bug fix: hope it works: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1554_publish.zip

alligator77, 2021-02-01:
> 
> 
> I made a quick bug fix: hope it works: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1554_publish.zip

Yes, its works! Thx very much!

closed 2021-02-06 by svenhb