svenhb/GRBL-Plotter #181: Gamepad cross-buttons support

alligator77, 2021-02-01:
Can You add gamepad cross-buttons support?

[image]

svenhb, 2021-02-02:
Are the cross keys different from keys 0 to 15?
[image]

alligator77, 2021-02-02:
> 
> 
> Are the cross keys different from keys 0 to 15?
> [image]

Looks like yes, don't light up when pressed.

alligator77, 2021-02-02:
I will add the description, check these buttons on the gamepad and what will be written in the line "Message from Gamepad" when I press these buttons.

svenhb, 2021-02-02:
Good idea, thanks

alligator77, 2021-02-02:
[image]

left : 27000
right : 9000
up : 0
down : 18000
neutral : -1

svenhb, 2021-02-04:
For each direction the first word is "PointOfViewControllers0"?
Unfortunatly my gamepad doesn't send this value, just Buttons and X,Y,Z,RotationZ.

alligator77, 2021-02-04:
> 
> 
> For each direction the first word is "PointOfViewControllers0"?
> Unfortunatly my gamepad doesn't send this value, just Buttons and X,Y,Z,RotationZ.

Yep, for all directions "PointOfViewControllers0".

svenhb, 2021-02-04:
And they should act like X,Y on full scale, or are individual commands ok, like for the buttons?

alligator77, 2021-02-04:
There are combinations of buttons, such as up and left. It is probably better to set jogging on all buttons with current feed or with adjustable feed.

alligator77, 2021-02-04:
For example to use them for more accurate supply to a point of

svenhb, 2021-02-06:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

alligator77, 2021-02-07:
It works! Tnx

svenhb, 2021-03-08:
Thanks for testing

closed 2021-03-08 by svenhb