svenhb/GRBL-Plotter #182: Speed Axis uncorrect when uses microstep divided.

78Matteo78, 2021-02-03:
Hello
I have a problem with the real axis speed since I use hardware divider microsteps.
from the control streaming menu the feed rate of the axes is not right. I ask if it is possible to put a menu that allows you to choose the hardware step divider in order to then correct the speed of the axes. Thank you
Greetings matteo

svenhb, 2021-02-04:
I don't get it: if you use a hardware divider it would half the speed, but also half the distance. After correcting the $100,$101 setting everything should be ok?

78Matteo78, 2021-02-04:
Hi
I'll explain my real example.
I have a 2mm pitch screw, a motor that does 200 steps per revolution, microsteps x8, so I set 800 mm steps on the parameter $ 100, 101,102.
I built a square with a side of 150 mm and a perimeter of 600 mm, the speed is set at 600 mm per minute. The square is perfect but the execution time is 2 minutes and not 1 minute. that's why I don't understand if there is a mistake in some steps or there is something in the software. we found the same thing in another cnc of a friend.
thanks bye.

svenhb, 2021-02-05:
When driving a square, the motors slow down and accelerate at the edges. So the parameters $ 120, $ 121 are important for acceleration. Perahps these values are too low and it takes a lot of time to decelerate and accelerate.
Feedrate is only valid after the acceleration of the motor.

78Matteo78, 2021-02-05:
Hi
120,121 are at 20,000 and it is very low as a value and does not affect the end of machining time.
I tried with another software and the result is always 2 minutes when it should be 1 minute.
Yet the length of the sides is 150mm .... I don't understand where the mistake can be.
I'll try to do other tests. 
greetings matteo.

78Matteo78, 2021-02-10:
Hi
I did some tests at different speeds, f450, f500, f600, f700, f900 and they all respect the travel times mm / min except f600 which took double the time traveling at half speed.
I also used other software and it had the same behavior.
Then I adjusted the speed from f600 by removing f1 in small steps and in the range f600 -f540 in addition to going half the speed, it also goes into lockout.
With other speeds there is no problem.
I don't understand if e is a mega grbl problem or a resonance problem ....
I'll try with an oscilloscope or frequency meter to see if I understand something.
Greetings matteo

svenhb, 2021-02-11:
I had a similar problem with differnet steps/mm settings:
$100 = 10 -> A command X1 results to a movement of 1.000
$101 = 4.444 -> A command Y1 results to a movement of 1.024
                          But a command Y1.000 results to a movement of 1.000
Strange...

svenhb, 2021-02-11:
Can you upload your grbl-setting ($-Values)? Would like to check if it also happens with my bare board on a desk.

78Matteo78, 2021-02-11:
ARDUINO MEGA 2560 5 AXIS

 $$
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=25          (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=0           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=1           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=1          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=0          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=0          (Enables homing cycle. Requires limit switches on all axes.)
< $23=0          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=25.000     (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=500.000    (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=1000.000   (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=255        (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=400.000   (X-axis travel resolution in steps per millimeter.)
< $101=400.000   (Y-axis travel resolution in steps per millimeter.)
< $102=400.000   (Z-axis travel resolution in steps per millimeter.)
< $103=13.300    (A-axis travel resolution in steps per degre.)
< $104=12.500    (B-axis travel resolution in steps per degre.)
< $110=900.000   (X-axis maximum rate. Used as G0 rapid rate.)
< $111=900.000   (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=500.000   (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=900.000   (A-axis maximum rate. Used as G0 rapid rate.)
< $114=200.000   (B-axis maximum rate. Used as G0 rapid rate.)
< $120=20.000    (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=20.000    (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=20.000    (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=100.000   (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $124=20.000    (B-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=20000.000 (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=20000.000 (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=20000.000 (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=20000.000 (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $134=20000.000 (Maximum B-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok

I HAVE NOT YET CHECKED WITH THE OSCILLOSCOPE ON ARDUINO MEGA, BUT ON ARDUINO 3 AXES IT HAS NO PROBLEMS, AND AT 600 MM / MIN THE PULSE FREQUENCY IS ABOUT 8 KHZ, WHAT I WOULD ALSO EXPECT ON MEGA.
AS SOON AS I CAN I WILL LET YOU KNOW.
GREETINGS MATTEO

78Matteo78, 2021-02-11:
MY DRIVER IS TB6600  VDC 32 VOLT AND MOTORS NEMA 24 200 STEPS

78Matteo78, 2021-02-13:
Hi
I tested with an oscilloscope the outputs of the steps in the output of arduino mega and as I imagined the frequency of the steps is halved, from F600 to about F540, or the data flow on usb is blocked .... I tried with other software (grblgru) and it had the same behavior so in my opinion there is a bug in the Mega firmware.
The firmware manager should be consulted.
Thanks again for the work.
Greetings Matteo.

svenhb, 2021-03-08:
As a reminder:
https://github.com/fra589/grbl-Mega-5X/issues/179

78Matteo78, 2021-03-13:
Hi svenhb

as you have seen I contacted the firmware programmer, but he too cannot give an explanation.
in fullstep the problem is not there, and in any case the machine works the same, for me the problem can be closed.
meanwhile I wanted to ask you if it is possible to give a direction and towards from inside to outside or vice versa, in the path of the gcode, when I create a simple shape.
Thank you
Greetings  #Matteo.

svenhb, 2021-03-14:
You may have an example? I don't understand exactly what you mean.

78Matteo78, 2021-03-14:
[image]

78Matteo78, 2021-03-14:
[image]

78Matteo78, 2021-03-14:
HELLO SVENHB
I HOPE YOU UNDERSTAND.

svenhb, 2021-03-15:
Yes, I understand, but what is it used for?

78Matteo78, 2021-03-16:
Hi
For example, I work with aluminum and therefore starting from the outside means being able to work with only a minimum part of the cutter and not at 100%.
Or in the reverse case where I had to widen a slot, start from the inside and work in small steps towards the outside with a maximum of 50% of the cutter.
I know maybe it's a bit complicated, they are functions that exist on artcam or aspire, but since creating work paths with grblplotter is much faster, it would be very nice.
If I complicate your life, do the same, don't worry ....
Thanks again
Greetings Matteo.