svenhb/GRBL-Plotter #183: Z axis servo reveresed

DFBRews, 2021-02-09:
Hello I am running the hex file for a plotter and the z axis servo movement is reversed I am very new to grbl and dont know how to proceed thanks

running an uno with v3 shield and have the signal for the servo into z + limit on the shield

svenhb, 2021-02-10:
Hi,
to use the servo you need to create specific gcode (commands to control your grbl-controller), which can be done by importing graphics (HPGL, SVG, DXF) in GRBL-Plotter with specific settings.
For servo, the yellow marked settings are important. To switch the up/down movement, just reverse the Pen-up / Pen-down values (in this case 200, 800) - which controles the servo position (possible range for this grbl-version is 0 to 1000 as I know).
[image]

More example settings you find here: https://grbl-plotter.de/index.php?id=setups&setlang=en

DFBRews, 2021-02-11:
Thanks for the reply Reversing the s800 and and s200 got the pen to move properly the issue I am having now is when the g code finish’s even though a pen up command is sent the pen stays down until I manually command it up 

Thanks 

Paul

svenhb, 2021-02-11:
You may have a short example code?
I assume the problem is a final switch off of the spindle (M5). If so, I should change this bevavior if using a Servo.
Perhaps you can add a M3 800 (pen up) in the footer section, to lift the pen finally.

svenhb, 2021-02-12:
The final M30 command is the reason, It switches off the spindle, this equals spindle speed = PWM value = 0.

svenhb, 2021-02-12:
Because i have the same problem at the moment, I made new version 1.5.6.1: https://github.com/svenhb/GRBL-Plotter/releases
The possibility to skip the M30 command

DFBRews, 2021-02-12:
thank you I will install it on Monday and see how it works and let you know. Thanks again for all the support

svenhb, 2021-02-27:
I added a new option - Check new version: https://github.com/svenhb/GRBL-Plotter/releases
Option to control the servo during setup of the Servo PWM values
[image]

DFBRews, 2021-03-01:
Thanks again for the help 
I am able to keep the pen up when the program is finished by removing the m30 completely from the footer and having the ignore m30 checked in the z servo box.

svenhb, 2021-03-08:
The option to avoid sending "M30" keeps the servo-position on it's last position and will not send a "M5", which causes the firmware to apply a pwm-value of 0: 
See https://github.com/cprezzi/grbl-servo/blob/eggbot/grbl/spindle_control.c comment:
// For servo send min. PWM instead of deactivate PWM

closed 2021-03-08 by svenhb