svenhb/GRBL-Plotter #184: Subrouting script gets wrong toolnumber.

hampijke, 2021-02-18:
Hello!
Facing some problem with subrouting script in grbl-plotter for a watercolorbot. So it starts from tool select correctly. picking color, but when it comes time to subrouting for brush refresh it picks a wrong color. Gcode seems to be correct with grouping and etc. (I might be wrong, quite new to plottering). Tried TOAX and TOLX all goes same. Is it some bug or just Im doing something wrong? It somehow feels inside subrouting it recieve a wrong tool number.
If there is need for more information just tell me what shall I send. 

[script_brush_OnlyRefreshColor_my_nc.txt](https://github.com/svenhb/GRBL-Plotter/files/6003384/script_brush_OnlyRefreshColor_my_nc.txt)

[GCODE_test_color_changer_.txt](https://github.com/svenhb/GRBL-Plotter/files/6003440/GCODE_test_color_changer_.txt)

Thanks in advance!

svenhb, 2021-02-19:
Probably I found the problem. I assume the subroutine uses coordinates of the last tool inside the gcode.
I will fix that and check....

hampijke, 2021-02-19:
> Probably I found the problem. I assume the subroutine uses coordinates of the last tool inside the gcode.
> I will fix that and check....

That would be great!

svenhb, 2021-02-20:
I made a new version and tested it... https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1562_publish.zip
Note the new "script_sr_brush..." scripts.
To make the scripts more variable, the pen-up/down code will be available as subroutines O97 and O98 - callable from the tool-change scripts. Then it's easier to switch from real Z movement to servo-control - what i used right now.
[image]

hampijke, 2021-02-21:
Nice, Im going to check it today! Thanks for that fast fix!

svenhb, 2021-02-27:
Found a bug, when using a real Z axis (not moving down the first time).

hampijke, 2021-02-27:
Yep. Thought its some sort of strange thing, but fixed it with using pen up/pen down gcode inserting, then it works.

svenhb, 2021-02-27:
Should be fixed now - Check new version: https://github.com/svenhb/GRBL-Plotter/releases

hampijke, 2021-03-03:
Seems to work fine!
One more question about watercolor. Is it possible to somehow make brush go down at the same time as the plotter starts painting the object, so that there wont be a dot under brush and the surface would be touched while movig? Could be quite usefull but now I think gcode is created by steps, so I cant make it move down and paint at same time or am I wrong?

svenhb, 2021-03-03:
That's a sensible idea. Yes at the moment the gcode will be built together with pen-down, path, pen-up, so the pen-down/up needs to be extended by an additional XY-move.
BTW: are you using the drag-tool compensation?

svenhb, 2021-03-03:
You may try this: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1572_publish.zip
Not really tested, hope you can do that.
[image]

hampijke, 2021-03-06:
> That's a sensible idea. Yes at the moment the gcode will be built together with pen-down, path, pen-up, so the pen-down/up needs to be extended by an additional XY-move.
> BTW: are you using the drag-tool compensation?

I was trying drag-tool but may be the vectors I used are not showing much of difference with drag tool on/off. Im gonna try more with it and may be post some reply here then. 
Thanks a lot for suggestion on Lead in/out it might be very usefull for whatercolor! Going to try it soon. Was a bit busy last week.

svenhb, 2021-03-08:
Initial problem seems to be solved, thanks for testing

closed 2021-03-08 by svenhb