svenhb/GRBL-Plotter #185: Updating SVG Fonts

x1project, 2021-02-20:
Hi Sven, first of all I really appreciate your work.
I am generally using "G-code Creation --> Create Text" section.
When I write some chars (ö ç i ü ğ ş) which are not part of standard chars, program gives empty space.
I know, those chars did not included in SVG Fonts which you took from Evil Mad Scientist, so empty space is normal.

But Windell Oskay updated his inkscape extension and SVG Font sets
Right now, when I use his inkscape extension, I can write (ö ç i ü ğ ş) chars in inkscape.

in issue #130 you mentioned "I just took the data I found"
I am kindly asking, could you use updated(extended) char set data (SVG Fonts) for next release?
This is the repository address https://gitlab.com/oskay/svg-fonts

Second sugestion;
in "Create G-code from Text" window, I can see LibraCad fonts in font section ("data\fonts\***")
is it possible to make some changes then when user copy/paste some SVG font file to "data\fonts\" 
then user can use that font? (I tried but didn't work)

svenhb, 2021-02-20:
Yes, good idea.
Probably next weekend...

svenhb, 2021-02-27:
Check new version: https://github.com/svenhb/GRBL-Plotter/releases 
[image]

closed 2021-03-08 by svenhb