svenhb/GRBL-Plotter #186: Repeat a file N times

amoineau, 2021-02-26:
**Context**
I don't know if you remember but I'm the person with the machine that cuts and marks para-glider wings, using a conveyor to feed the fabric.
Everything is going well with the project but I recently realized I was going to have a small problem. We can make a set of wings, using the tiling and 3rd serial port systems you implemented, but after that we need to manually start the process to get a second set.
The purpose of the project being to automate the production of wings, we would like to be able to launch a series of N wings and come back after a few hours.

**Request**
With the current implementation of grbl-plotter, is it possible to do something like that ? If not, would it be possible to have a field near the "play" button to choose how many time we want the file to be executed ?

**Considered solutions**
I noticed the "path repetition (for weak laser)" in the setup, but I am not sure it is compatible with all the "exotic" options we are using. For instance, it doesn't seem to replicate the footer code but I need it to send a command to the conveyor in order to push the last tile of the previous set and get fresh fabric for the new set.  
Most importantly, the person we are making the machine for probably won't appreciate a solution requiring to tinker in the setup before every program, that's why having the option on the main form would be great !

Thank you for your help.
Antoine.

svenhb, 2021-02-27:
Hope this liitle change is enough - Check new version: https://github.com/svenhb/GRBL-Plotter/releases 
[image]

amoineau, 2021-03-01:
Thanks, I'll try it.

amoineau, 2021-03-04:
With the latest update a "subroutine" block appeared in my G-Code. I tried to find why that is, but I couldn't. Do you have the answer ?
Otherwise the "repeat all" seems to work fine. I'll do further testing this very afternoon, I'll get back to you.

svenhb, 2021-03-04:
Bug fix in Tool change script handling #184
To make the scripts more variable, the pen-up/down code will be available as subroutines O97 and O98 - callable from the tool-change scripts. Then it's easier to switch from real Z movement to servo-control.

Then you don't need to change external scripts if you change Pen-up/down behavior.

amoineau, 2021-03-04:
Ok so I can just ignore it ?

svenhb, 2021-03-04:
Yes, and if you don't need the subroutines you could remove this code...

amoineau, 2021-03-04:
What should I remove exactly ? Not that it is a big problem but since it is repeated by the "Repeat code > All" feature, it can become a bit cluttering.

svenhb, 2021-03-04:
Just the O97 / O98 blocks...
I will add an option to avoid the subroutines

amoineau, 2021-03-04:
Oh you meant in the gcode directly. Sorry, did not get that.
Ok thanks !

amoineau, 2021-03-04:
So... Apparently it's not working that well.
* As the program was streaming I kept having this error pops up :

[image]

After I clicked "OK" it proposed to exit GRBL-plotter, I refused and it kept doing that regularly.
Strangely it did not seem to pose problem, the program did not stop and went through.

* It might not be related, but many lines in the gcode were highlighted in yellow in a seemingly random pattern.

All of this was on a DXF file that I've been using to run tests for quite a while.

amoineau, 2021-03-05:
Another thing I noticed (maybe it's not related to the repeat feature at all and maybe it occurred before without me noticing) :  

[image]

I got that while trying to make a really long file (5m), 3 times, for a total of 15 tiles (1 tile = 1m on the machine). I never tried that file before so I don't know if the "problem" comes from a recent change. And perhaps it's not even a problem.

svenhb, 2021-03-05:
What exactly do you mean? "Too less values to group"? If just one item is found, which was selected for grouping, then there is nothing to group - beacuse of problems whith that, I then disable grouping.

Or do you mean the yellow color? This marks the edit-mode, which disables automatic marking of figures or groups. This can be enabled/disabled on right-click on editor.

amoineau, 2021-03-05:
Oh yes sorry I was so in my head I forgot to precise ! yes I was talking about "to less values to group"

amoineau, 2021-03-05:
Ok so It's not a problem and I don't have anything to do ?

svenhb, 2021-03-05:
I think it's not a problem - except you expect two tools (Aren't you using laser and pen?)

amoineau, 2021-03-05:
Not on this file, I was working more on the tile / file repetition. I just had one long oval with the laser.

amoineau, 2021-03-05:
But on the file that caused the thread exception above I was using both tools on a 3 tiles file, repeated twice.

svenhb, 2021-03-08:
Error in line 186 of MainFormStreaming should be fixed: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2021-03-09:
I'll give it a try this afternoon, thank you !

svenhb, 2021-03-09:
Note:
Improovement in Tool change script handling #184
Subroutines O97 and O98 will only be generated if subroutine calls are found in tool-change scripts.

amoineau, 2021-03-09:
wow great !

amoineau, 2021-03-18:
I've been using it for some time and I believe this issue can be closed now

closed 2021-03-18 by amoineau

svenhb, 2021-03-18:
Thanks for testing