svenhb/GRBL-Plotter #187: Possibility to fix height differences of the machine axis'

metzger100, 2021-02-26:
My selfbuilt CNC is not perfectly aligned. So the right side of my basepalte is 1,5mm lower than the left side.
After the machine is a Sprite-typed machine I can't fix this difference of height on the machine without completly rebuilding it. Maybe it is possible to fix in the software. It would be awesome.

svenhb, 2021-02-26:
You may use the height map option (Menu Workpiece - Surface Scan/Height map) :
1. Option you measure the map of your machine bed and apply it to yout workiece, or
2. You create the map by your self and load it into the Surface-scan dialog.
Height mao file looks like this (MaxX/Y = dimension in mm, SizeX/Y = number of points):
```
<?xml version="1.0" encoding="utf-8"?>
<heightmap MinX="0" MinY="0" MaxX="100" MaxY="100" SizeX="2" SizeY="2">
  <point X="0" Y="0">0</point>
  <point X="0" Y="1">0</point>
  <point X="1" Y="0">-1</point>
  <point X="1" Y="1">-1</point>
</heightmap>
```
This results to (just watch color map and use blue button):
[image]

Only problem: the heigth map is bound to the work coordinate 0;0 and not to the machine coordinate.

metzger100, 2021-02-27:
Yes. I already use the heightmap, but it would the life of all selfmade CNC-men easier if you could implement that Solution i proposed above. I would love to see that in a future version.
Greetings.

svenhb, 2021-02-27:
You wrote "After the machine is a Sprite-typed machine I can't fix this difference of height on the machine"
What doesit mean "sprite-typed machine", do you have a picture?

metzger100, 2021-02-27:
Sure. Maybe you know grblgru? The dev uses those CNC-types for his Gcode creation. The types describe the relations between workpiece and the axis of the machine. e.g Workpiece is fixed, Z axis is fixed, co.
Spritetyp machines have the workpiece on the y-axis and the z-axis is mounted on the x-axis.
[image]

svenhb, 2021-02-27:
I just knew the german wording...
[image]

metzger100, 2021-02-27:
Ah okay. 
Ja. Ich kenn nur die Typen auf Englisch. Hab im Studium auch nur die englischen Begriffe gehabt, aber das scheint das selbe zu sein.
Danke für deine Arbeit.
Hab nachdem ich nun auch selbst ohne Gegenleistung Software anzubieten hab ich noch mehr Respekt vor deiner Leistung.
Vielen Dank. Sehr cool!

svenhb, 2021-03-15:
Probably it's easier to use a waste board, where you mill off the surface with fixed Z value - then the surface is parallel to your XY mechanincs and you can fix your workpiece on it. Thats the way other CNC user do it...

metzger100, 2021-03-16:
Thats how how I am doing right now. Didn't think about it when I opened the request.

closed 2021-03-16 by metzger100

svenhb, 2021-03-16:
Thanks for the feedback