svenhb/GRBL-Plotter #189: In the same dxf file, choose a use case that is not used, and the empty stroke arrows are displayed differently

lianzaozi, 2021-04-01:
image1 uses example_laser.ini, the arrow direction of the empty stroke is correct
[image]

[image]

image2 uses example_plotter.ini, the arrow direction of the empty stroke is wrong
[image]
[image]

Test dxf file
[2008.zip](https://github.com/svenhb/GRBL-Plotter/files/6241503/2008.zip)

svenhb, 2021-04-06:
Thanks, for the detailed explanation, the problem should be fixed here: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1577_publish.zip

lianzaozi, 2021-04-13:
I downloaded the latest version of the software for testing, and found that except for the direction of the arrow on the first track is incorrect, everything else is correct.

[image]
[image]

[image]
[image]

svenhb, 2021-05-02:
I think it's fixed

closed 2021-05-02 by svenhb