svenhb/GRBL-Plotter #190: The latest version of the software text filling cannot be used

lianzaozi, 2021-04-15:
The latest version of the software text filling cannot be used.

The software text filling function of version 1.5.0.1 is normal.

svenhb, 2021-04-15:
I didn't changed this function - for me it's working:
[image]

Also vector filling for SVG is working, DXF-fill is not implemented.
[image]

[SVG_Layers_filled.zip](https://github.com/svenhb/GRBL-Plotter/files/6315558/SVG_Layers_filled.zip)

lianzaozi, 2021-04-16:
This is my operation process, the software version I use is 1.5.7.8
[image]

svenhb, 2021-05-02:
You have to use smaller values for this numbers because your dimension are smaller:
[image]

closed 2021-07-31 by svenhb