svenhb/GRBL-Plotter #192: Is it possible to add split line display and set coincidence line length function to the cutting function?

lianzaozi, 2021-04-16:
Hello, the cutting function in the software is very easy to use at present. I hope to add an option that can control whether to display the dividing line and numbers that control the processing sequence of the displayed segmented area. Now I find that the effect of laser processing at the dividing boundary is not very good. The solution is the previous processing The end point of the processing path in the area should exceed the dividing line by a little bit, and the processing start point of the next area should also exceed the dividing line by a little bit, so that the processing effect is better.
I hope to add an option in the software to allow users to set a parameter that can determine the repeat length of the split path.

split line display

[image]

Repeat length

[image]

Example of user setting 1mm repeat length
The current movement mode of the software is A=>E, E=>D; A is the starting point of the straight line of the last divided area, E is the end point of the straight line of the last area; E is the starting point of the straight line of the next divided area, and D is the next area The end of the straight line

If the repetition distance is set to 1mm, the movement mode is A=>C, B=>D; A is the starting point of the straight line of the last divided area, C is the end point of the straight line of the previous area after the repetitive distance is set; B is the next divided area The starting point of the straight line, D is the end point of the straight line in the next area after the repeat distance is set

svenhb, 2021-04-18:
Please check: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1579_publish.zip

lianzaozi, 2021-04-26:
Thank you, I will test some tests. I downloaded the software source code of version V1.5.8.0, but when I compiled in Visual Studio, I was wrong.

closed 2021-07-31 by svenhb