svenhb/GRBL-Plotter #194: Can some simple drawing functions be integrated?

lianzaozi, 2021-04-23:
Can some simple drawing functions be integrated? Such as commonly used points, circles, arcs, straight lines, ellipses, etc.
There are already some codes that can be used for reference, such as: Canvas, SharpCAD, LitCAD 

SharpCAD (https://github.com/zhudaijie/SharpCAD)
LitCAD (https://github.com/wangyao1052/LitCAD)
Canvas
[C# CAD.zip](https://github.com/svenhb/GRBL-Plotter/files/6364313/C.CAD.zip)

svenhb, 2021-04-23:
Did you already noticed the jog path editor?
It was made to create quickly a path in jog-mode e.g. to flaten a surface:
E.g. set "Raster" with tool radius and draw a meander shape in the dimensions of your workpiece surface...
[image]

svenhb, 2021-04-23:
I think it is more convenient to use a familiar program - like InkScape, than learning a new one and missing some features...
Did you know: you can copy objects out of InkScape and paste them into GRBL-Plotter. I think this is far more practical.

lianzaozi, 2021-04-26:
understand what you mean.

I have this idea is to take into account some simple graphics I don't have to use third-party software, and can be done directly with the plotter software. And the individual geometric primitives in the figure is also more convenient, such as changing a line size, angle, changing the radius of a garden, and so on. And save these changes in the map, which is convenient for next time.

svenhb, 2022-05-28:
Now also nodes of a figure can be moved.
You may have geometric primitives as SVG in a folder and add them to GRBL-Plotter, when needed, than change geometry with given functions in GRBL-Plotter: move, scale, rotate...

closed 2022-05-28 by svenhb