svenhb/GRBL-Plotter #195: HEX File for CoreXY plotters available for GRBL-plotter?

Keepthefunk, 2021-04-23:
I can find .hex files for the Arduino that seem work fine with the GRBL Plotter Software - but my system is CoreXY. 

Is there any coreXY hex file I can download and flash on to my LY drawing robot? I love the grbl-plotter software and would like to use it with my CoreXY plotter.
This is the one I would like to control with grbl-plotter:

https://de.aliexpress.com/item/33005228398.html?spm=a2g0o.productlist.0.0.dfbf7ab76ttF05&algo_pvid=6262d7e5-10ed-4f32-87d6-5328efe71b6f&algo_expid=6262d7e5-10ed-4f32-87d6-5328efe71b6f-2&btsid=2100bdd016192113081967803ee6f6&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

svenhb, 2021-04-23:
I don't have coreXY hardware, so there was no need to build such a grbl version and I can't test the firmware...
Did you try it by yourself? It's not so difficult...
Start here to get grbl compiled and uploaded: https://github.com/gnea/grbl/wiki/Compiling-Grbl
Edit config.h to enable coreXY (line 189) and read here: https://github.com/grbl/grbl/issues/996
If you need servo support you should take the files from here: https://github.com/cprezzi/grbl-servo

Perhaps here you find a ready to use hex file inside the zip file - in section Compiled Firmware HEX: 
 https://www.electrodragon.com/w/Arduino_CNC 

If it is running as expected I can put your hex file to the GRBL-Plotter setup if you like.

svenhb, 2021-04-23:
After reading your post a 2nd time:
How do you control your plotter right now? Perhaps the firmware is already grbl? What is the start-up message of the controller?
Do you have schematic to know pin assignment for step and dir signals?

Keepthefunk, 2021-04-24:
Thanks! First thing I tried out the HEX files from electrodragon but they are zipped and password protected :-|
I will try the source code from cprezzi then.
I have this cheap plotter that came with with GRBL 0.9. It works fine but the software that came with it is terrible. I would love to use your software. But using grbl plotter my plotter freezes after 20-30 minutes and the servo gets hot. 
When I use your hex files the servo works just fine! Only the stepper motors move in the wrong direction because my plotter is corexy.
I think a lot of people would benefit from a corexy hex file. Thanks again!

Keepthefunk, 2021-04-24:
heh, the password I guessed now

Keepthefunk, 2021-04-25:
At last my plotter works fine with your software! I went with the small angle hex file:
https://www.electrodragon.com/w/File:V1.1_GRBL_coreXY_servo_servo_small_angle.zip
And my corexy plotter works now even with complex jobs that run overnight. 
Here is a review of this machine:
https://luruke.medium.com/chinese-penplotter-at-55-aka-cnc-v3-shield-toys-diy-ly-drawbot-pen-lettering-corexy-xy-plotter-82f79b28dea1
The hardware is fine but the software only works with GCode and no resizig or positioning. And mine froze in the middle. Not any more.
Thanks for your help!

Keepthefunk, 2021-04-25:
closed because solved for me - thanks again!

closed 2021-04-25 by Keepthefunk

svenhb, 2021-04-25:
Perhaps you can give a hint to the password of the zip? So I can extract the hex file and add it to my collection :-)

Keepthefunk, 2021-04-26:
Yes sure - password is electrodragon. The HEX files there are fine. 

But oh dear - I still have these freezes after sometimes 30 minutes and sometimes up to three hours. 
I have replaced: 
- The Uno (and even flashed new firmware with your help)
- The controller software (grbl plotter)
- The servo
- The pololu drivers
- The USB Cable
I can still replace the CNC V3 Shield and see what that does.

I am beginning to think my windows laptop causes this. Checked USB power settings and turned power saving off. Still freezing. Will try a desktop computer next.
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/6379762/logfile.txt)

Any help is greatly appreciated. Where can I seek the error?

svenhb, 2021-04-26:
What exactly happens? The logfile.txt is very short, without any error - did you just upload the last lines? Do you get any error message? 
Perhaps a memory issue inside the laptop?

Keepthefunk, 2021-04-26:
Thanks for looking in to this. Here is another example - it just froze again. 
My laptop works fine with another GRBL laser cutter but I will have to replace that also for troubleshooting. 
[logfile (1).txt](https://github.com/svenhb/GRBL-Plotter/files/6380045/logfile.1.txt)

svenhb, 2021-04-26:
I think it is USB related, the program looses connection...
Perhaps worn-out USB connector causes loose contact. Or broken USB cable?

Keepthefunk, 2021-04-28:
Thanks! The fact that I don't respond quickly is not because I don't care but more because searching for the error takes long. 
Thanks for the hint with USB issues - I had better results with my trusty desktop but it also failed after a few hours. . Maybe I have more than one error here and I have to sort out two. Now with quality USB cable to be sure. 
Also, I am reinstalling the CH341 driver and trying that out.
I really have to get into grbl compiling myself so I tweak the baud rate to 9600 and try that. Will keep you updated!

svenhb, 2021-04-29:
Perhaps it is grbl-setup related: if $100 steps/mm is very high and you use then high feed rate, you exceed the 30kHz step-signal limit (for Arduino uno). This also causes some strange behavior...
You may press this button to check this:
[image]

Do you use special hardware or can you quickly replace the board (Arduino nano / uno)?

svenhb, 2021-04-29:
BTW: https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-serial-transmission-errors

Keepthefunk, 2021-04-29:
My $100 is 100. Here are my GRBL settings. Nothing extraordinary, right?
[GRBL Settings.txt](https://github.com/svenhb/GRBL-Plotter/files/6399884/GRBL.Settings.txt)
No special hardware here. So I just replaced the elegoo Uno with a new AZdelivery UNO as in the picture.
The serial driver I did find on https://ftdichip.com/drivers/vcp-drivers/ 

I am now running a new test. I hope this one goes through!
[image]
[image]

Keepthefunk, 2021-04-29:
OK it froze again with the new UNO. I also tried swapping the CNC shield with another one - also freezes after one hour. 
Now I have exchanged every piece of hardware and am out of ideas

svenhb, 2021-04-29:
You may check the logfile (same location as the exe). Any error at the end of the file?
Perhaps power saving mode of the PC? (Should be disabled automaticly by GRBL-Plotter during a run)

svenhb, 2021-04-29:
Whats about the power supply (of the plotter)? Perhaps it shuts down because of overheating?

Keepthefunk, 2021-04-29:
The original power supply had only 12V 2A so I was suspicious about that. I replaced it with a 5A version.  Does not even get warm. But to no avail. I tried both. 
Here is the protocol of the latest freeze
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/6401858/logfile.txt)
I will have to think about what to do next. Thanks anyway so far.

svenhb, 2021-04-30:
Strange: again the logfile just ends without any hint...
Did you kill GRBL-Plotter via Task-Manager?

svenhb, 2021-04-30:
>to my Makelangelo wall plotter   
 
this one you also control via GRBL-Plotter? What is the grbl-version?

The used grbl version is 0.9 (is this the electrodragon version?) 
Perhaps you should try to edit the actual grbl release to coreXY (https://github.com/gnea/grbl)

Strange errors inside the logfile:
```
2021-04-30 00:17:02.9809 | Warn  | GRBL_Plotter.ControlSerialForm | Ser:1  grbl ERROR 'error: Invalid gcode ID:24'  no information found for error-nr. 'Invalid gcode ID'
2021-04-30 00:17:03.0434 | Warn  | GRBL_Plotter.ControlSerialForm | ! Last RX 'G01 X148.775 Y175.544'  BufferFree: 17  Index:29675  max:29684  lineNr:29687
 
 
2021-04-30 02:16:47.2363 | Error | GRBL_Plotter.MainForm          | fCTBCodeMarkLine fCTBCodeClickedLineLast:195729 fCTBCodeClickedLineNow:195729 System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   at FastColoredTextBoxNS.Bookmarks.Contains(Int32 lineIndex)
   at FastColoredTextBoxNS.FastColoredTextBox.BookmarkLine(Int32 iLine)
   at GRBL_Plotter.MainForm.fCTBCodeMarkLine(Boolean markAnyway)
```

kptjrn, 2021-05-02:
Hi Keepthefunk,

I'm also running a Core XY machine, and have uploaded a hex-file for these in issue #199 (which has default settings for my machine (Change these in GRBL-Plotter with "$=")
It is not strictly necessary to use a hex-file, as you can also just upload GRBL with the Arduino IDE (which also means you can change the settings and default files for you system)

You might have tried this, but regarding your issues with pauses and stops it sounds a bit like an issue I've had before, where the Stepper Drivers on the CNC-Shield got too hot, which made the motors stutter or stop. Have you tried adjusting the Current Setting on the drivers (The tiny potentiometer next to the heat-sink)? If this is set too high, the drivers will get too hot, and your system looks very light weight, so shouldn't require very much power.

svenhb, 2021-05-02:
You may try coreXY version, based on grbl 1.1h: https://github.com/svenhb/GRBL-Plotter/issues/199
But it is probably without rc-servo support.

Keepthefunk, 2021-05-03:
Hi all, thanks a lot for your compassion. After a few more days of trying I found the culprit: It was the servo cable - one single dupont plug on the signal pin was not crimped right. Wow! 
And the overlying problem is that the g-code I am testing with seems to have some issues and brings the port to crash after some time. Can this be possible? 
I attached one of the test jobs I left running overnight and this one still not goes through.
[Kahlo Portrait.zip](https://github.com/svenhb/GRBL-Plotter/files/6417407/Kahlo.Portrait.zip)
This is a CMYK picture for four different pens with M0 commands to make the plotter wait for the next pen color.

OK in the meantime I have learnt to compile my own GRBL. That is really not hard as you mentioned. Didn't know this is found as archive in IDE. Now I know. I think next thing for me is to learn more about g-code and why my g-code does not go though. 

So, time to clean up my workplace and take a walk. Thanks!