svenhb/GRBL-Plotter #196: zprobe touches block stops then continues down

tcpd555, 2021-04-26:
when i use the probing x and y probe works fine, then i do the z probe and it comes down hits block stops for a millisecond and then speeds up down into the block, i have used the probe on UGS and works fine. its just on the z that it does this
[image]

svenhb, 2021-04-27:
Strange... Did you proceed in on go ("Probe Z also" checked). Or first XY and after that "Only Z"?

tcpd555, 2021-04-27:
i tried it 3 ways, xy with z probe also and bottom left xy, and z only , z probe comes down, touches block very briefly stops then speeds up and drives into block, i have a piece of aluminum rod in spindle now so i dont snap any more tips on bits to test it with

tcpd555, 2021-04-27:
here is a video of the zprobe 

[](url)https://www.youtube.com/watch?v=ApWSABQG_cE

svenhb, 2021-04-27:
Wow really bad...
I think the problem is: the "Save distance Z" is smaller than the "Touch plate dimension Z".
After successful Z-Probing, the save distance is 10 mm above the wood, but unfortunatly 11.9 mm below your touch plate...
I should take account of the touch plate dimensions: final position = save distance + touch plate dimension

[image]

tcpd555, 2021-04-27:
that was it, i changed save distance to 32 and worked fine, thank you

svenhb, 2021-04-27:
Very good.
I have thought about it once again: I think it's better to adapt "Save distance" value, when "Touch plate dimension" exceeds this value. 
Because I would expect "Save distance" above workpiece and not above "Touch plate" which I remove after probing.

"Thanks" for finding this issue - too bad you looses some bits.

closed 2021-07-31 by svenhb