svenhb/GRBL-Plotter #197: Add another feature please

mmcmNew, 2021-04-28:
When creating a g code from the text, so that the movement is not on the outer contour, but on the inner one.

svenhb, 2021-04-28:
Can you make a picture? I don't get the problem...
The GCode is created on the contour....
[image]

You could use radius compensation to get the outer contour of closed shapes:
If the contour is on the wrong side, try negative value for radius
[image]

mmcmNew, 2021-04-28:
[image]

But I need inner one

svenhb, 2021-04-28:
If this is the original font, you can't remove it.
You may try radius compensation with negativ value
[image]

svenhb, 2021-04-28:
With this font and characters it works:
[image]

mmcmNew, 2021-04-28:
This doesn't work with all fonts, but thank you.

mmcmNew, 2021-04-28:
I ran into another problem. When you send gcode for execution at the beginning, it is written well, but towards the end it starts to be executed in jerks. But if you press the pause button in the program at this moment, the execution will continue smoothly again

https://user-images.githubusercontent.com/77157663/116414000-c61b8e00-a840-11eb-9fc0-2e0b6cdc02eb.mp4

svenhb, 2021-04-28:
Did you enable this checkboxes? This may cause delays...
[image]

mmcmNew, 2021-04-28:
not enabled

mmcmNew, 2021-04-28:
Enabled
This happens at the end of the code. If the text is large, then it will be on the extreme lines

svenhb, 2021-04-28:
Perhaps a memory problem...
You may watch Streaming progress - the green bars and Task-manager memory usage...
What kind of PC and operating system do you use?

mmcmNew, 2021-04-28:
Win 10. Laptop
vimple.co/93a4f3c1e6a24efdb08d300bc1426657

svenhb, 2021-04-28:
I think your feedrate is too high and the grbl buffer runs empty. With pause there is chance to fill up the grbl buffer again.
What controller do you use?
Same problem in my video, check at 4:40 https://youtu.be/7Nvf4UwK_yk?t=282

mmcmNew, 2021-04-28:
It's may be. Using USB Type-C port

mmcmNew, 2021-04-28:
Is it possible to implement the ability to specify the page width when creating code from text, and thus the text would automatically be transferred to the next line? Otherwise, you have to put hyphenation manually throughout the text

svenhb, 2021-04-28:
It is grbl related, somewhere I read that at maximum 10 Gcode lines per second can be processed to fill up the internal path algorythm...

svenhb, 2021-04-28:
> the ability to specify the page width  
  
I need to check this

mmcmNew, 2021-04-28:
it will be very convenient

mmcmNew, 2021-04-28:
thank you so much for your help

mmcmNew, 2021-04-28:
https://youtu.be/wuy6Ry_xmRU?t=107
And the ability to trace along the center line too I think will be very convenient especially when writing with a pen

mmcmNew, 2021-04-29:
I'm tired of trying to get the printer to write what I need
[image]
[image]
since there is no creation of code from the font along the center line, I created an image without a background through Corel. The result is shown in the screenshot. I don't understand how to do it normally.
In addition, whatever image I insert, it also creates a frame along the borders of this image, but I don't need it

svenhb, 2021-04-29:
> The result is shown in the screenshot. I don't understand how to do it normally. 
 
Strange idea - why don't use another font? For me it works:
[image]

Settings - I pressed "Comic - few colors" reduced number of colors to 2, increased brightness, use small ResoX:
[image]
[image]
[image]

>In addition, whatever image I insert, it also creates a frame along the borders of this image, but I don't need it  

It just shows the graphic dimension, it is no gcode wich will be processed... You can switch off here:
[image]

svenhb, 2021-05-02:
Check new version for automatic line break on given page width... https://github.com/svenhb/GRBL-Plotter/releases
[image]

mmcmNew, 2021-05-02:
Thanks) This is very convenient. 
Is there a chance that a centerline will appear for the text?)

svenhb, 2021-05-02:
What do you mean with center line?

mmcmNew, 2021-05-02:
https://youtu.be/wuy6Ry_xmRU?t=113

mmcmNew, 2021-05-02:
So that the text is created not on the contour, but on the center line

svenhb, 2021-05-02:
Sorry, I think this is too complicted for me

mmcmNew, 2021-05-03:
What procedure is responsible for drawing letters to the program?
I'll try to figure it out and do what I need to do

svenhb, 2021-05-03:
Text will be created from LFF of SVG fonts: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/GCodeCreation/CreateFromForm check for files with 'text' or 'font' in the name.
The example above - finding inside path - is imaging processing - same path, files with 'image' in the name. 
 
What's about the font you used - from where did you get it? Perhaps there is also the style available you need?
[image]
 

My suggestion is: make your own SVG font - here are some links to start with:
https://www.evilmadscientist.com/2019/hershey-text-v30/ 
https://inkscape-manuals.readthedocs.io/en/latest/creating-custom-fonts.html

mmcmNew, 2021-05-04:
I want to try to do this so as not to redo the fonts. I think it will be easier this way

svenhb, 2021-05-04:
You may try it in inkscape:
https://inkscape.org/de/~arpruss/%E2%98%85centerline-trace-on-windows-instructions

closed 2021-07-31 by svenhb