svenhb/GRBL-Plotter #198: MKS DLC 2.0  firmware DLC-CNC

qulet, 2021-04-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

qulet, 2021-04-30:
I don't want to connect to serial port.
"GRBL Panel" connects correctly.
I don't know where the problem is.
This occurs in V1.8, I had an older version of the GRBL plotter and it was not a problem.
Where can I download the older version of the program from?

svenhb, 2021-04-30:
I never heared about this MKS DLC....
What kind of error do you get? Any screenshots?

About older  version - just scroll down here: https://github.com/svenhb/GRBL-Plotter/releases

qulet, 2021-04-30:
Thank you for your response
[image]

qulet, 2021-04-30:
MKS DLC V2
[MKS DLC V2.0_001 SCH.pdf](https://github.com/svenhb/GRBL-Plotter/files/6407775/MKS.DLC.V2.0_001.SCH.pdf)
[MKS DLC V2.0_001 PIN.pdf](https://github.com/svenhb/GRBL-Plotter/files/6407778/MKS.DLC.V2.0_001.PIN.pdf)

qulet, 2021-04-30:
Pin_out  arduino-UNO

svenhb, 2021-04-30:
You wrote GRBL Panel connects to your hardware - you should check the selecteded baudrate. I assume the default baudrate (115200) is not correct for your board:
[image]

qulet, 2021-05-02:
115200 is the correct value with this speed my MKS board connects to programs "GRBL Panel" and "Ligthtburn"

svenhb, 2021-05-02:
Can you send the logfile (same folder as the exe)?

svenhb, 2021-05-02:
Do you get any welcome message, when using the other programs? I assume your firmware behaves different compared to "classic" grbl.

qulet, 2021-05-02:
Hello
The problem occurs when I run applications from a file:
GRBL-Plotter.application

disappears when I run the program from a file:
GRBL-Plotter.exe
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/6411730/logfile.txt)

qulet, 2021-05-02:
Thank you very much for your help and for providing such versatile software !!!
The problem is clarified, I close the topic
Greetings !

qulet, 2021-05-02:
[logfile.2.txt](https://github.com/svenhb/GRBL-Plotter/files/6411794/logfile.2.txt)

Unfortunately, the problem has reappeared. What's with this buffer?
[image]
[image]

svenhb, 2021-05-02:
Please try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1581_publish.zip
To be able to switch off the alarm via $X - Kill Alarm.

Buffer size is the size of the serial receive buffer inside the Arduino  Uno (Arduino Mega has more buffer...)

qulet, 2021-05-02:
GRBL-Plotter_1581_publish.zip
GRBL-Plotter_1582_publish.zip
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/6412003/logfile.txt)
homing not work....

svenhb, 2021-05-02:
You need to correct your settings, grbl message:
`grbl ALARM 'ALARM:9' Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring.`

qulet, 2021-05-11:
svenhb Please contact me privately by e-mail
kuleta@wtvk.pl

For help in starting the ATtiny44 pen lifter

closed 2021-07-31 by svenhb