svenhb/GRBL-Plotter #199: Hi, thanks for an interesting piece of software:)

kptjrn, 2021-05-01:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kptjrn, 2021-05-01:
Hi, thanks for an interesting piece of software :)

I have previously used Inkscape Gcode-tools to run my DIY Plotters, but there seems to be no support for that anymore.
Your GRBL-Plotter looks very promising, but I can't seem to get my plotter and the software to connect right.
I have uploaded the GRBL firmware myself (and not through the Xloader), because I need some settings changed for running a Core XY machine.
Can that be the problem?

Using GRBP-Plotter 1.5.8.0
I can connect but get no movement or control of the machine.
Tested using UGS (Universal Gcode Sender and there was no problem with moving
Also, there a red alarm button on, that I cannot remove, and get this from the "COM CNC":

> [CTRL-X] reset
Form will be reduced after connecting
* RESET
< Grbl 1.1h ['$' for help]
* Version 1.x
* Read grbl settings, hide response from '$$', '$#'
* Assume buffer size of 127 bytes
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS|WCO:-840.000,-670.000,6.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:RHS|Ov:100,100,100>

Any help would be greatly appreciated :-D
Best, Jørn

svenhb, 2021-05-01:
Up to now I don't have a coreXY machine, but in principle grbl should work with GBRL-Plotter... 
About the Alarm - I found a bug: $X to cancel the alarm will not be sent, I fixed this....
Please check the update: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1581_publish.zip

Would be nice if you could show your coreXY Hex file... 
Another user has problems with v 0.9: https://github.com/svenhb/GRBL-Plotter/issues/195

kptjrn, 2021-05-02:
Thank you Sven, that worked perfectly :)

I've attached the hex-file I use, and I verified it by uploading it with the Xloader, and all seems fine. I have also attached the GRBL file with modifications for my system:

 _- Changes to config.h for running Core XY
 - and defaults.h changed to fit my machine (all the things you can change in in GRBL-Plotter with "$ = ??")_

There is still an issue with the Core XY setup, which will not let you home the Z-axis. It's a bit annoying, but I can live with it. (This has nothing to do with your software, it's a GRBL thing)

If you run the attached firmware on a regular XY plotter you will get som very erratic behavior, where the machine moves diagonally instead of straight.

I hope it is for some use :)

--
Btw, I get this error message when running the setup (but it doesn't seem to affect anything?):

_"Fehler beim Herunterladen der folgenden Ressource:
http://go.microsoft.com/fwlink/?linkid=164199&clcid=0x411"

Datum: 2021-05-02 13:28:51

Weitere Informationen finden Sie in der Setup-Protokolldatei unter "C:\Users\KPTJRN~1\AppData\Local\Temp\VSD6317.tmp\install.log".
"_

[image]

[grblUpload.ino.zip](https://github.com/svenhb/GRBL-Plotter/files/6411359/grblUpload.ino.zip)

[grbl-1.1h.20190825.zip](https://github.com/svenhb/GRBL-Plotter/files/6411361/grbl-1.1h.20190825.zip)

svenhb, 2021-05-02:
Thanks for the feedback, 
About the installation-error: perhaps you can show the mentioned logfile: C:\Users\KPTJRN~1\AppData\Local\Temp\VSD6317.tmp\install.log

kptjrn, 2021-05-02:
Sure, it's here

[install.log](https://github.com/svenhb/GRBL-Plotter/files/6411509/install.log)

And thank you for the help :)

svenhb, 2021-05-02:
It seems, that some DOT.NET files were missing and the setup-tool tried to download them: 
```
Downloading files to "C:\Users\KPTJRN~1\AppData\Local\Temp\VSD6317.tmp\"
(2021-05-02 13:28:48) Downloading 'DotNetFX40KB2468871\dotNetFx40LP_Full_x86_x64de.exe' from 'http://go.microsoft.com/fwlink/?linkid=164199&clcid=0x411' to 'C:\Users\KPTJRN~1\AppData\Local\Temp\VSD6317.tmp\'
Download completed at 2021-05-02 13:28:51
Downloading failed with HRESULT=-2146697210
```

Perhaps windows tries to get the files with the next update?
Thanks for the feedback!

closed 2021-05-02 by svenhb