svenhb/GRBL-Plotter #200: The input box is blocked by the window, and the text cannot be input

wwwlxs, 2021-05-09:
**Describe the bug**
The input box is blocked by the window, and the text cannot be input.Can't scroll the page at the same time

**Screenshots**

[image]

svenhb, 2021-05-09:
I don't know why the layout is so different, In my form designer, the apply button has the same Y coordinate as the Reload button, but on your system not...
Your screenshot: 
[image]
My form designer: 
[image]

svenhb, 2021-05-18:
I reordered the options, please try: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1583_publish.zip

svenhb, 2021-05-19:
Perhaps you can send a screenshot to show if the button "Apply changes" is still moved upwards

svenhb, 2021-07-31:
I cleaned up the window

closed 2021-07-31 by svenhb