svenhb/GRBL-Plotter #201: Spline Bugs

Behiiitaj, 2021-05-18:
**Spline bugs**
Hi...
Sorry, I'm not good in English...
I open a 'Dxf' file that contain spline. (DXF file exported from "Solidworks" )
Unfortunately 'GRBL_plotter' generate g_code like this...
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File.'
2. Click on 'Open file'
3. Select the 'example_router.ini' use case
4. See error

svenhb, 2021-05-18:
Try to increase this number in setup:
[image]

svenhb, 2021-05-18:
And be sure this one is not checked:
[image]

Behiiitaj, 2021-05-18:
This is my file

svenhb, 2021-05-18:
No attachment...

Behiiitaj, 2021-05-18:
https://drive.google.com/file/d/131l8l16vk3T2ImMVjilJMBomSrM_H022/view?usp=drivesdk

svenhb, 2021-05-18:
Ok, I get the same result - I will try to fix it

svenhb, 2021-05-18:
Found and fixed the problem, please try: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1583_publish.zip

Behiiitaj, 2021-05-19:
Thank you so much
I've tested the program, The problem has fixed.

Behiiitaj, 2021-05-23:
Is source code updated after fixing spline problem?

svenhb, 2021-05-23:
I need some time to practical check the other improovements I built in (more positions for rc-servo, available as subroutines for improoved watercolor usage). But at the moment my hobby room is occupied by other things.

svenhb, 2021-07-31:
Source code is updated

closed 2021-07-31 by svenhb