svenhb/GRBL-Plotter #202: Detecting tools from layers

amoineau, 2021-06-02:
**Describe the bug**
Not sure if it's actually a bug, maybe I'm just missing something.
I use these options to detect which tool the machine is supposed to use :

[image]

The tool table looks something like that : 

[image]

And the detection is not consistent. I can have a file with an empty default layer and a second layer called LASER with the forms to cut and GRBL-P will pair the layer with the ENCRE tool. 

I recently had the idea to match the color as well and it seems to be much more consistent. 
If it works like that it's not a huge problem but it makes creating the DXFs and tool tables a bit more tedious and prone to error.

svenhb, 2021-06-02:
Did you select the desired properties here?
[image]

svenhb, 2021-06-02:
Perhaps differences in uppercase lowercase? I use the LevenshteinDistanceAlgorithm to judge tool-name versus layer-name.
I guess "LASER" gives the same score as "encre" if you look for "ENCRE" (5 chars needs to be replaced).

amoineau, 2021-06-02:
Hi, yes I made sure the case was the same on both sides and both tools. 
I did select "XY feedrate from TT" and "Spindle Speed from TT" but not the Z axis one. Although I don't use a Z axis and this window is different on my version (1.5.8.2):
[image]

(Am I using too many decimal places ?)

amoineau, 2021-06-03:
> Did you saved your new tool-table?

[laser_and_pen_ISP.zip](https://github.com/svenhb/GRBL-Plotter/files/6589336/laser_and_pen_ISP.zip)

> Can you try different layer-name length?

Sure, I can try. Do you mean that each layer should have a different length than the others, or that the layer name should be longer ?

> I can't check the code before weekend.

No problem ! Thank you for your help.

svenhb, 2021-06-03:
I don't know why: in your tool table the seperators are not consistent. All my examples uses "," comma, but yours also shows ";" semicolons - could be a problem.
Different name-length would be good as a test: yes: Do you mean that each layer should have a different length than the others.

amoineau, 2021-06-03:
If you mean the last column, it's a separator for the commands to execute when switching to the tool, not a csv separator.
I played around with different name length and casing and it didn't solve the problem, but matching the colours does.

One file where this problem appears :  
[ia21a1342pmi0002-01_slc.zip](https://github.com/svenhb/GRBL-Plotter/files/6590225/ia21a1342pmi0002-01_slc.zip)

svenhb, 2021-06-03:
I think the problem is, that this DXF file has only one layer with objects.
Because of a problem in earlier time #147, grouping with just one group-item will not be performed...

I need to check for a workarround

amoineau, 2021-06-03:
I thought so because there are information in the g-code header about that, so I tried earlier this week to add an empty ENCRE layer on another version of this file and it didn't change anything :(

svenhb, 2021-06-06:
Hope it is fixed: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2021-06-07:
Thx, I'll try it this week !

closed 2021-07-31 by svenhb