svenhb/GRBL-Plotter #203: Problem using the Gamepad

mrthekod, 2021-06-03:
I used to have a gamepad control working wonderfully... Then I moved house.
Now I just get this. 
Tried different gamepads/joysticks/Arduino Leonardo pretending to be a gamepad. All the same.
Everything else is working as expected.
[image]

mrthekod, 2021-06-03:
My current Gamepad Settings.
[image]

svenhb, 2021-06-03:
1st screenshot: you send G91 Y0.000 F0 -> F0 means no speed at all.
Are you using grbl version 1.1 or 0.9?
Does the virtual joystick works?
[image]

mrthekod, 2021-06-03:
Virtual Joystick works fine.
The inputs report fine in gamepad settings, i.e. Go from - 0x0 to 0xFFFF
Tried using older versions of GRBL, currently back on 1.1
Tried 6 other versions of GRBL-Sender.
Even tried a whole new install of Win10!!
Most frustrating.

mrthekod, 2021-06-03:
Currently Flashing the arduino nano3 with grbl_v1.1h.20190825.hex that was downloaded with GRBL-Plotter.

svenhb, 2021-06-03:
Do you get always the same command send, as visibe on your screenshot? (G91 Y0.000 F0 ) - I mean F0

svenhb, 2021-06-03:
Also the step with is zero (Y0.000) - strange

svenhb, 2021-06-03:
What are your grbl settings for $100, $101?

mrthekod, 2021-06-03:
I was getting about 100 a second.

Update!
Deleted EVERYTHING, started from scratch with 1.5.8.2 and the grbl_v1.1h.20190825.hex that was in the 1.5.8.2 zip.

All of a sudden, works perfectly! 
The only thing I really did differently, was to use  Xloader to send the HEX file. Usually used the Arduino ide.

Thankyou for your attention tho. Will return if anything odd happens.

I do have a bit of an odd project coming up soon with a VERY elaborate tool change system. Should I use this or an email if I hit a snag... probably won't be a bug so much.

svenhb, 2021-06-03:
>Should I use this or an email  

I would prefer this way

svenhb, 2021-06-03:
BTW: there is already a newer version "nonreleased" with some bug fixes: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1583_publish.zip

mrthekod, 2021-06-03:
Oh I'm not messing with this for a while :)
Thankyou again.

closed 2021-06-03 by mrthekod