svenhb/GRBL-Plotter #205: instalation error

hottec2003, 2021-06-13:
when i try to install i get a error

Error downloading the following resource:
http://go.microsoft.com/fwlink/?linkid=164199&clcid=0x411

svenhb, 2021-06-13:
I assume the installer tries to get dot-net 4.0 from microsoft, but the link doesn't work.
Perhaps you can try to install it manually, before running the GRBL-Plotter setup: https://www.microsoft.com/en-US/download/details.aspx?id=17718

StuartRolph, 2021-06-20:
I have the same issue. I downloaded the dot-net 4.0 installer but it says that it is already installed on my system. The GRBL-plotter installer still gives the same error.

svenhb, 2021-06-20:
I found this, I hope this helps: "However, if some programs require you to install the lower version (4.0) and you are unable to install it, you may try using .Net framework repair tool to repair the currently installed packages and then enable the framework version 4.0. If the repair tool does not work, you may try the .Net framework cleanup tool to remove the installed versions of .Net framework and then reinstall them manually. Please refer to the instructions given in the link below."

From: https://answers.microsoft.com/en-us/windows/forum/windows_10-other_settings/net-40-will-not-load/0bc02212-4c13-4e42-8e4a-e5789b365803

StuartRolph, 2021-06-20:
I tried all the suggestions in that post and several others, yet I still have the same issue on multiple computers I have tried. I even tried disabling .NET 4.8 in Windows Features and I was still unable to run the .NET 4.0 installer. It seems that since it is now built into Windows 10 it won't allow for separate installations. I am curious why an older .NET version is required since it should be backwards compatible.

svenhb, 2021-06-21:
Thats strange... 
The reason for use of .NET 4.0 is the support of windows XP - which I still have on my old control-notebooks. 
You may try to install an other grbl-sender sw, which uses also .NET 4.0: https://github.com/arkypita/LaserGRBL to see if this works, perhaps I then can find the problem. 

I found more information here: https://docs.microsoft.com/en-us/dotnet/framework/install/on-windows-xp

Kinzakindzadza, 2021-06-27:
I tried to install Version 1.5.8.4 and older version of GRBL-Plotter on Win10 with .NET v. 4.8 and on WinXP with .NET v. 4.0.30319. There are similar errors when setup is starting. Same time, LaserGRBL installed successfully on these computers. May be the problem is in something else?
Would be grateful for your help!

svenhb, 2021-06-27:
>Same time, LaserGRBL installed successfully

Thanks for this information. 
The errors are still about .NET 4.0? Perhaps you can show a screenshot of the error?

svenhb, 2021-06-27:
BTW: the source code contains still the subfolder "GRBL-Plotter\bin\Release" which contains all files and the exe. Just start "GRBL-Plotter\bin\Release\GRBL-Plotter.exe" at least this should work - I hope.

svenhb, 2021-07-31:
Check new setup of pre-release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2021-07-31 by svenhb