svenhb/GRBL-Plotter #206: Step movement function

lianzaozi, 2021-06-27:
Does the plotter software have the function of step movement?
For example: I set 10mm, and every time I click the button, the axis moves 10mm.

svenhb, 2021-06-27:
Sure, just uncheck "STOP jogging on mouse up", then the full move wil be perfomed. If you click 3 times, it moves 3*distance. 
If checked, the move will be interrupted, when rleasing the mouse:
[image]

lianzaozi, 2021-06-27:
Thank you, I tried it and it was very easy to use.

but how to adjust the manual speed of the axis?

lianzaozi, 2021-06-27:
Chinese translation correction

The correct Chinese translation of "Stop jogging on mouse up" is "Stop jogging on mouse up"

svenhb, 2021-06-27:
Thanks for correction.
Setup of jogging is here:
"Calculation of best speed" is not correct (a bit too fast). 10 mm with feedrate F=1000 is ok, calculation gives F=1200, which causes interrupted moves if mouse is pressed for a longer time.
[image]

lianzaozi, 2021-06-27:
thank you

The source code of version 1.5.8.4 fails to compile in the visual studio of my computer, and an error will be reported.

svenhb, 2021-06-27:
>The source code of version 1.5.8.4 fails to compile  
 
I don't know how I could help

lianzaozi, 2021-06-27:
You can download the software source code of V1.5.8.4, open it with VS on your computer, and compile it to see if there is an error.

svenhb, 2021-06-27:
>You can download the software source code of V1.5.8.4, open it with VS on your computer, and compile it to see if there is an error.
 
No errors, just warnings... Perhaps you can show a screenshot of the error?
Perhaps it is a missing nuget packet... I thought just delivering the DLLs is enough - perhaps not:
[image]

lianzaozi, 2021-06-27:
My operating software steps are like this:
[image]

[image]

svenhb, 2021-06-27:
Perhaps you need to disable some debug options? 
What happens if you select "Release":
[image]

lianzaozi, 2021-06-27:
Thank you for your reply, you are right.

The problem disappeared after selecting "Release", but I don't understand why there is a problem in regenerating the source code after selecting "debug"

lianzaozi, 2021-06-27:
After selecting "debug" on your computer, will there be problems in regenerating the source code?

svenhb, 2021-06-27:
In debug mode, stronger rules will be applied. I don't know how to setup. If I run it in Debug mode I get no errors...

svenhb, 2021-06-27:
Sometime you can disable warnings, if you click on them. Don't know if it will work with this links:
[image]

svenhb, 2021-06-27:
Perhaps if you now switch back to "Debug" it works?

lianzaozi, 2021-06-27:
The problem is still not solved. Clicking on the error message link will jump to the Microsoft website. This page explains that the source code needs to be modified to solve the error.

lianzaozi, 2021-06-28:
try rebuild solution

svenhb, 2021-06-29:
Perhaps this is the trick:
[image]

lianzaozi, 2021-06-30:
My computer set
[image]

svenhb, 2021-07-01:
Perhaps you need to select another rule or reset the rule: https://stackoverflow.com/questions/33637211/vs2015-warning-msb3884-could-not-find-rule-set-file

lianzaozi, 2021-07-01:
Thank you for your reply

I kept the same settings of visual studio unchanged, then opened the source code of another program, clicked to regenerate the solution, and no error was reported.
I think the error reported when compiling the source code of the plotter should have nothing to do with the visual studio settings.

svenhb, 2021-07-01:
>I think the error reported when compiling the source code of the plotter should have nothing to do with the visual studio settings.

But it has: if you select "Release" you get no errors because of different rules...

lianzaozi, 2021-07-01:
I copied some wrong links for vs jump, these pages explain the cause of the error.
I have the location of these error codes in the screenshot above.

CA1500
https://docs.microsoft.com/zh-cn/previous-versions/visualstudio/visual-studio-2015/code-quality/ca1500-variable-names-should-not-match-field-names?view=vs-2015&redirectedfrom=MSDN

CA1502
https://docs.microsoft.com/zh-cn/previous-versions/visualstudio/visual-studio-2015/code-quality/ca1502-avoid-excessive-complexity?view=vs-2015&redirectedfrom=MSDN

CA1806
https://docs.microsoft.com/zh-cn/previous-versions/visualstudio/visual-studio-2015/code-quality/ca1806-do-not-ignore-method-results?view=vs-2015&redirectedfrom=MSDN

CA1505
https://docs.microsoft.com/zh-cn/previous-versions/visualstudio/visual-studio-2015/code-quality/ca1505-avoid-unmaintainable-code?view=vs-2015&redirectedfrom=MSDN

svenhb, 2021-07-01:
Yes, but these are warnings - as the webpages says, not errors. Your setup is not ok...

lianzaozi, 2021-07-01:
I reset the settings of visual studio and restarted visual studio, but the problem is still not resolved
[image]

[image]

svenhb, 2021-07-01:
Did you try VS 2019 or VS 2022? Both are also working...
Did you load the project from the GRBL-Plotter.sln file? Perhaps thats the differnece?

lianzaozi, 2021-07-01:
My computer only has VS 2019 installed.
I load the project from the "GRBL-Plotter.sln file".

svenhb, 2021-07-01:
Whats about this setting: Treat warnings as errors:
[image]

lianzaozi, 2021-07-02:
Thank you for your reply.

The settings of the vs software on my computer are the same as in your screenshot.

lianzaozi, 2021-07-03:
I tested other programs can compile normally

lianzaozi, 2021-07-06:
VS2022 compilation code failed, error code CA0001, this is the only error

svenhb, 2021-07-06:
I'm reworking the code, to get rid of most warnings - give me some weeks...
Just another try for your problem:
Close project in visual studio
Perhaps it helps if you replace with text-editor in GRBL-Plotter.csproj:
2nd line "<Project ToolsVersion="14.0" ..." by
<Project ToolsVersion="4.0" 
Open project

svenhb, 2021-07-31:
Check new pre-release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2021-07-31 by svenhb