svenhb/GRBL-Plotter #207: 这个刀具补偿在哪啊

lirui-dot, 2021-07-22:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lirui-dot, 2021-07-22:
Where is tool compensation

svenhb, 2021-07-22:
Do you mean tool-length compensation? 
Check here: https://grbl-plotter.de/index.php?id=form-probe&setlang=en

lirui-dot, 2021-07-22:
> 你的意思是刀具长度补偿？
> 在这里查看：[https](https://grbl-plotter.de/index.php?id=form-probe&setlang=en) : [//grbl-plotter.de/index.php?id=form-probe&setlang=en](https://grbl-plotter.de/index.php?id=form-probe&setlang=en)

是要怎么设置呢？

svenhb, 2021-07-22:
In this video you may get an idea: https://youtu.be/x5UTHpgsfII
You need a fix switch (connected to grbl -probe pin) to detect the tool Z position.

lirui-dot, 2021-07-22:
可以加一个改变G代码的吗？

lirui-dot, 2021-07-22:
> 在此视频中，您可能会有一个想法：[https](https://youtu.be/x5UTHpgsfII) :
> //youtu.be/x5UTHpgsfII 您需要一个固定开关（连接到 grbl -probe 引脚）来检测工具 Z 位置。

就是想有一个设置刀具直径改变图形的位置

svenhb, 2021-07-22:
You can add the code for tool length compensation to the tool change scripts.
Example without tool-length compenastion: https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en
GCode for probing is G91 G38.3 Z-10 if you want to move 10 mm down.
G43.1 Zx will then set the tool length.
Better read here: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38
https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g43.1

lirui-dot, 2021-07-22:
> 您可以将刀具长度补偿的代码添加到换刀脚本中。
> 没有工具长度补偿的示例：[https:](https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en)
> //grbl-plotter.de/index.php [?](https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en) id [=](https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en) auto -pen-change [&](https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en) setlang [=](https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en) en 如果您想向下移动 10 毫米，则用于探测的 GCode 为 G91 G38.3 Z-10。
> G43.1 Zx 将设置刀具长度。
> 最好在这里阅读：[http](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38) : [//linuxcnc.org/docs/html/gcode/g-code.html#gcode](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38) : [g38 ](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38)
> [https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g43 .1](https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g43.1)

我就是想改变G01的坐标，根据他的刀具长度来修改

svenhb, 2021-07-22:
Then follow the instruction: https://grbl-plotter.de/index.php?id=form-probe&setlang=en

Tool length correction is a bit tricky:

- 1st get the tool length of the first tool 
- Then zero the Z position via G92 as needed 
- Start work 
- After a tool change, measure tool length of new tool

lirui-dot, 2021-07-22:
> 然后按照说明操作：[https](https://grbl-plotter.de/index.php?id=form-probe&setlang=en) : [//grbl-plotter.de/index.php?id=form-probe&setlang=en](https://grbl-plotter.de/index.php?id=form-probe&setlang=en)
> 
> 刀具长度修正有点棘手：
> 
> * 1st 获取第一个刀具的刀具长度
> * 然后根据需要通过 G92 将 Z 位置归零
> * 开始工作
> * 换刀后，测量新刀具的刀具长度

根据补偿刀具半径来改变G01的XY的坐标

svenhb, 2021-07-22:
You mean cutter radius compensation?
Then try this:
[image]

lirui-dot, 2021-07-22:
> 你是说刀具半径补偿？
> 然后试试这个：
> [image]

这个设置了没有什么用啊

svenhb, 2021-07-22:
Then I can't help

lirui-dot, 2021-07-23:
> 那我无能为力

铣削时上刀具半径补偿是要怎么计算呢，怎么偏移所有图形的刀具补偿R呢？

svenhb, 2021-07-23:
[image]
Raidus compensation only works on XY axis.
If no figure is selected, all figures will be transformed.
Radius compensation may fail on complicated shapes.

lirui-dot, 2021-07-26:
> [image]
> 半径补偿仅适用于 XY 轴。
> 如果未选择任何图形，则将转换所有图形。
> 在复杂的形状上，半径补偿可能会失败。

 是要先设置好，在进行加载文件吗？

lirui-dot, 2021-07-26:
> > [image]
> > 公式推定仅
> > 适用于 XY 轴。如果未选择任何图形，则将转换所有图形。
> > 在模拟的形状上，证明可能会失败。
> 
> 是要先设置好，在进行加载文件吗？

我现在设置了，没有反应，是我设置错了吗？

lirui-dot, 2021-07-26:
[image]

lirui-dot, 2021-07-26:
[image]

> > > [image]
> > > 公式推定仅
> > > 适用于 XY 轴。如果未选择任何图形，则将转换所有图形。
> > > 在模拟的形状上，证明可能会失败。
> > 
> > 
> > 是要先设置好，在进行加载文件吗？
> 
> 我现在设置了，没有反应，是我设置错了吗？

[image]

lirui-dot, 2021-07-26:
> > > [image]
> > > 公式推定仅
> > > 适用于 XY 轴。如果未选择任何图形，则将转换所有图形。
> > > 在模拟的形状上，证明可能会失败。
> > 
> > 
> > 是要先设置好，在进行加载文件吗？
> 
> 我现在设置了，没有反应，是我设置错了吗？

[image]

svenhb, 2021-07-26:
If the drawing is visible, select radius compensation, put in the value (e.g. +10 for outside or -10 for inside the shape) and press enter: light gray shows the previous shape.
[image]

lirui-dot, 2021-07-26:
> 如果图形是可见的，选择半径补偿，输入数值（例如+10 表示外部或-10 表示内部）并按回车键：浅灰色显示之前的形状。
> [image]

[image]
这种图形不可以吗

lirui-dot, 2021-07-26:
> > 如果图形是可见的，选择半径补偿，输入数值（例如+10 表示外部或-10 表示内部）并按回车键：浅灰色显示之前的形状。
> > [image]
> 
> [image]
> 这种图形不可以吗

[image]
这样是什么原因呢，图形复杂了吗？

lirui-dot, 2021-07-26:
> > > 如果图形是可见的，选择半径补偿，输入数值（例如+10 表示外部或-10 表示内部）并按回车键：浅灰色显示之前的形状。
> > > [image]
> > 
> > 
> > [image]
> > 这种图形不可以吗
> 
> [image]
> 这样是什么原因呢，图形复杂了吗？

我设置的是外部，为什么有一些图形是往内部的呢

svenhb, 2021-07-26:
Compensation inside / outside depends on drawing direction (clockwise or counter clockwise).
Please upload gcode, so I can check for the problem

lirui-dot, 2021-07-26:
> 内部/外部补偿取决于绘制方向（顺时针或逆时针）。
> 请上传gcode，以便我检查问题

是发文件给你吗？

lirui-dot, 2021-07-26:
> > 内部/外部补偿取决于绘制方向（顺时针或逆时针）。
> > 请上传gcode，以便我检查问题
> 
> 是发文件给你吗？

[image]
ha还有圆都是被删除了

svenhb, 2021-07-26:
You can drop the file here (perhaps make a zip-file before) 
With my test data, circles are working:
[image]

lirui-dot, 2021-07-26:
> 你可以把文件放在这里（也许之前制作一个 zip 文件）
> 使用我的测试数据，圆圈正在工作：
> [image]
怎么操作啊

svenhb, 2021-07-26:
Also your DXF works:
Perhaps you use some strange import options? Can you also upload the content of the editor?
[image]

lirui-dot, 2021-07-26:
> 您的 DXF 也有效：
> 也许您使用了一些奇怪的导入选项？也可以上传编辑器的内容吗？
> [image]

是上传生成的G代码吗

lirui-dot, 2021-07-26:
( Use case: ? )
( DXF import by GRBL-Plotter )
( Source: D:\康纯医疗\Drawing1.dxf )
(<Header >)
( G-Code lines: 121 )
( Pen Down/Up : 19 times )
( Duration ca.: 13.7 min. )
( Conv. time  : 00:00:00.0850411 )
( Layer: 0 . color: 7 . line type: Continuous)
(</Header >)
F1000 (Setup - GCode-Header)
G90
(<Figure Id="1" Geometry="Circle" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="314.2"> )
G00 X650.000 Y1100.000  
G02 X650.000 Y1100.000  I-50.000 J0.000 F1000 
(</Figure>)
(<Figure Id="2" Geometry="Polyline" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="480.0"> )
G00 X750.000 Y1150.000  
G01 X890.000 Y1150.000 F1000 
G01 X890.000 Y1050.000  
G01 X750.000 Y1050.000  
G01 X750.000 Y1150.000  
G00 X1050.000 Y1160.000  
G01 X992.937 Y1118.541 F1000 
G01 X1014.733 Y1051.459  
G01 X1085.267 Y1051.459  
G01 X1107.063 Y1118.541  
G01 X1050.000 Y1160.000  
(</Figure>)
(<Figure Id="3" Geometry="Line" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="323.6"> )
G00 X1250.000 Y1150.000  
G01 X1200.000 Y1050.000 F1000 
G01 X1300.000 Y1050.000  
G01 X1250.000 Y1150.000  
(</Figure>)
(<Figure Id="4" Geometry="Spline" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="421.6"> )
G00 X1410.000 Y1100.000  
G01 X1415.854 Y1104.972 F1000 
G01 X1421.741 Y1109.875  
G01 X1427.697 Y1114.641  
G01 X1433.756 Y1119.200  
G01 X1439.950 Y1123.484  
G01 X1446.316 Y1127.425  
G01 X1452.886 Y1130.953  
G01 X1459.695 Y1134.000  
G01 X1466.776 Y1136.497  
G01 X1474.165 Y1138.375  
G01 X1481.895 Y1139.566  
G01 X1490.000 Y1140.000  
G01 X1499.793 Y1139.508  
G01 X1509.907 Y1138.021  
G01 X1520.121 Y1135.637  
G01 X1530.213 Y1132.454  
G01 X1539.962 Y1128.571  
G01 X1549.146 Y1124.084  
G01 X1557.545 Y1119.093  
G01 X1564.936 Y1113.695  
G01 X1571.098 Y1107.989  
G01 X1575.811 Y1102.072  
G01 X1578.852 Y1096.043  
G01 X1580.000 Y1090.000  
G01 X1579.106 Y1084.034  
G01 X1576.309 Y1078.207  
G01 X1571.818 Y1072.577  
G01 X1565.846 Y1067.199  
G01 X1558.603 Y1062.129  
G01 X1550.299 Y1057.424  
G01 X1541.144 Y1053.141  
G01 X1531.351 Y1049.334  
G01 X1521.129 Y1046.060  
G01 X1510.690 Y1043.376  
G01 X1500.243 Y1041.337  
G01 X1490.000 Y1040.000  
G01 X1485.065 Y1039.615  
G01 X1480.240 Y1039.425  
G01 X1475.534 Y1039.436  
G01 X1470.956 Y1039.658  
G01 X1466.515 Y1040.099  
G01 X1462.220 Y1040.767  
G01 X1458.079 Y1041.670  
G01 X1454.101 Y1042.817  
G01 X1450.296 Y1044.215  
G01 X1446.671 Y1045.872  
G01 X1443.236 Y1047.798  
G01 X1440.000 Y1050.000  
G01 X1436.548 Y1052.860  
G01 X1433.344 Y1056.061  
G01 X1430.365 Y1059.571  
G01 X1427.591 Y1063.361  
G01 X1424.997 Y1067.398  
G01 X1422.561 Y1071.653  
G01 X1420.260 Y1076.093  
G01 X1418.073 Y1080.689  
G01 X1415.975 Y1085.408  
G01 X1413.946 Y1090.220  
G01 X1411.962 Y1095.095  
G01 X1410.000 Y1100.000  
(</Figure>)
(<Figure Id="5" Geometry="Arc" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="91.3"> )
G00 X1738.411 Y1102.009  
G03 X1659.313 Y1099.062  I-38.411 J-32.009 F1000 
G00 X1723.047 Y1089.206  
G03 X1675.588 Y1087.437  I-23.047 J-19.206 F1000 
G01 X1659.313 Y1099.062  
(</Figure>)
(<Figure Id="6" Geometry="Line" PenColor="00FF00" PenWidth="0.00" Layer="0" PathLength="20.0"> )
G00 X1723.047 Y1089.206  
G01 X1738.411 Y1102.009 F1000 
(</Figure>)
(<Figure Id="7" Geometry="Circle" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="314.2"> )
G00 X650.000 Y700.000  
G02 X650.000 Y700.000  I-50.000 J0.000 F1000 
(</Figure>)
(<Figure Id="8" Geometry="Polyline" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="480.0"> )
G00 X750.000 Y750.000  
G01 X890.000 Y750.000 F1000 
G01 X890.000 Y650.000  
G01 X750.000 Y650.000  
G01 X750.000 Y750.000  
(</Figure>)
(<Figure Id="9" Geometry="Circle" PenColor="FF0000" PenWidth="0.00" Layer="0" PathLength="314.2"> )
G00 X650.000 Y870.000  
G02 X650.000 Y870.000  I-50.000 J0.000 F1000 
(</Figure>)
(<Figure Id="10" Geometry="Polyline" PenColor="FF0000" PenWidth="0.00" Layer="0" PathLength="480.0"> )
G00 X750.000 Y920.000  
G01 X890.000 Y920.000 F1000 
G01 X890.000 Y820.000  
G01 X750.000 Y820.000  
G01 X750.000 Y920.000  
(</Figure>)
(<Figure Id="11" Geometry="Polyline" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="4220.0"> )
G00 X510.000 Y1250.000  
G01 X1830.000 Y1250.000 F1000 
G01 X1830.000 Y460.000  
G01 X510.000 Y460.000  
G01 X510.000 Y1250.000  
(</Figure>)
(<Figure Id="12" Geometry="Circle" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="478.5"> )
G00 X1086.158 Y920.000  
G02 X1086.158 Y920.000  I-76.158 J0.000 F1000 
G00 X1478.995 Y830.000  
G02 X1478.995 Y830.000  I-98.995 J0.000 F1000 
(</Figure>)
(<Figure Id="13" Geometry="Circle" PenColor="FF0000" PenWidth="0.00" Layer="0" PathLength="579.3"> )
G00 X1152.195 Y710.000  
G02 X1152.195 Y710.000  I-92.195 J0.000 F1000 
G00 X1704.403 Y850.000  
G02 X1704.403 Y850.000  I-104.403 J0.000 F1000 
G00 X1632.426 Y650.000  
G02 X1632.426 Y650.000  I-42.426 J0.000 F1000 
(</Figure>)
(<Figure Id="14" Geometry="Circle" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="177.7"> )
G00 X1478.284 Y600.000  
G02 X1478.284 Y600.000  I-28.284 J0.000 F1000 
(</Figure>)
G0X0Y0 (Setup - GCode-Footer)
M30

lirui-dot, 2021-07-26:
> > 您的 DXF 也有效：
> > 也许您使用了一些奇怪的导入选项？也可以上传编辑器的内容吗？
> > [image]
> 
> 是上传生成的G代码吗

[image]
这个为什么是往里面的，是我的图形的原因吗

svenhb, 2021-07-26:
Because some figures are clockwise, others are counter-clockwise drawn. Radius compensation follows the path and shifts the path by the given offset.

lirui-dot, 2021-07-26:
> 因为有些图形是顺时针绘制的，有些图形是逆时针绘制的。半径补偿跟随路径并将路径移动给定的偏移量。
It doesn't matter

svenhb, 2021-07-26:
After switching the direction in LibreCad, it works:
[image]
[image]

lirui-dot, 2021-07-26:
> 在 LibreCad 中切换方向后，它工作：
> [image]
> [image]

你这里偏移设置的是半径吗

svenhb, 2021-07-26:
[image]

lirui-dot, 2021-07-26:
> > 在 LibreCad 中切换方向后，它工作：
> > [image]
> > [image]
> 
> 你这里偏移设置的是半径吗

我知道了，谢谢作者，感谢感谢

lirui-dot, 2021-07-26:
> > > 在 LibreCad 中切换方向后，它工作：
> > > [image]
> > > [image]
> > 
> > 
> > 你这里偏移设置的是半径吗
> 
> 我知道了，谢谢作者，感谢感谢

这个地方如何获取圆的直径呢？

svenhb, 2021-07-26:
I don't understand your question - sorry

lirui-dot, 2021-07-26:
> 我不明白你的问题 - 抱歉

> I don't understand your question - sorry

If I want to know the radius of the circle, how do I set it

svenhb, 2021-07-26:
The radius of which circle? In which program?

lirui-dot, 2021-07-26:
> The radius of which circle? In which program?

[image]
The radius of this circle, how do we know its radius

svenhb, 2021-07-26:
How did you make the drawing? At least your CAD program can tell you the radius. 
If you click a figure, the corresponding gcode will be shown and you could read the radius from the G02 or G03  I and J parameters 

```
(<Figure Id="12" Geometry="Circle" PenColor="FF0000" PenWidth="0.00" Layer="0" PathLength="579.3"> )
G00 X642.195 Y250.000  
G01 Z-2.000 F1000 (PD)
G02 X642.195 Y250.000  I-92.195 J0.000 F2000 
G00 Z2.000 (PU)
G00 X1194.403 Y390.000  
G01 Z-2.000 F1000 (PD)
G02 X1194.403 Y390.000  I-104.403 J0.000 F2000 
G00 Z2.000 (PU)
G00 X1122.426 Y190.000  
G01 Z-2.000 F1000 (PD)
G02 X1122.426 Y190.000  I-42.426 J0.000 F2000 
G00 Z2.000 (PU)
(</Figure>)
```

lirui-dot, 2021-07-26:
> 你是怎么画图的？至少您的 CAD 程序可以告诉您半径。
> 如果你点击一个图形，会显示相应的 gcode，你可以从 G02 或 G03 I 和 J 参数中读取半径
> 
> ```
> (<Figure Id="12" Geometry="Circle" PenColor="FF0000" PenWidth="0.00" Layer="0" PathLength="579.3"> )
> G00 X642.195 Y250.000  
> G01 Z-2.000 F1000 (PD)
> G02 X642.195 Y250.000  I-92.195 J0.000 F2000 
> G00 Z2.000 (PU)
> G00 X1194.403 Y390.000  
> G01 Z-2.000 F1000 (PD)
> G02 X1194.403 Y390.000  I-104.403 J0.000 F2000 
> G00 Z2.000 (PU)
> G00 X1122.426 Y190.000  
> G01 Z-2.000 F1000 (PD)
> G02 X1122.426 Y190.000  I-42.426 J0.000 F2000 
> G00 Z2.000 (PU)
> (</Figure>)
> ```

Oh, thank you

lirui-dot, 2021-07-27:
> Oh, thank you

[image]
Where do I need to set it to get the right radius I for him

lirui-dot, 2021-07-27:
> [image]
> Where do I need to set it to get the right radius I for him

The radius here is 3. He's wrong here

lirui-dot, 2021-07-27:
> The radius here is 3. He's wrong here

[222.txt](https://github.com/svenhb/GRBL-Plotter/files/6882543/222.txt)
This is my DXF file

lirui-dot, 2021-07-27:
> [222.txt](https://github.com/svenhb/GRBL-Plotter/files/6882543/222.txt)
> This is my DXF file

[image]
Or can I set this green line to be all the way to the right of the circle?

lirui-dot, 2021-07-27:
> [image]
> Or can I set this green line to be all the way to the right of the circle?

Still draw cad graph to set according to you

svenhb, 2021-07-27:
>The radius here is 3. He's wrong here 

In GCode r°2 = I°2 + J°2
#
>Or can I set this green line to be all the way to the right of the circle? 

Just switch off this option, which changes the start point within a closed figure, for shortest connections:
[image]

lirui-dot, 2021-07-27:
> > The radius here is 3. He's wrong here
> 
> In GCode r°2 = I°2 + J°2
> 
> > Or can I set this green line to be all the way to the right of the circle?
> 
> Just switch off this option, which changes the start point within a closed figure, for shortest connections:
> [image]

Ok, thank you

lirui-dot, 2021-07-28:
> > > The radius here is 3. He's wrong here
> > 
> > 
> > In GCode r°2 = I°2 + J°2
> > > Or can I set this green line to be all the way to the right of the circle?
> > 
> > 
> > Just switch off this option, which changes the start point within a closed figure, for shortest connections:
> > [image]
> 
> Ok, thank you

[image]
I changed the background color, how do I set it to recognize that the color is white
[image]

lirui-dot, 2021-07-28:
> [image]
> I changed the background color, how do I set it to recognize that the color is white
> [image]

Or where can I change the background color

svenhb, 2021-07-28:
>I changed the background color, how do I set it to recognize that the color is white 

It's for Import: in LibreCad default is white pen on black background - but you don't wan't to plot with a white pen. For this I have the option use white as black. 

If you want to change the color inside the 2-D view, look here:
[image]

lirui-dot, 2021-07-28:
> > 我更改了背景颜色，如何设置才能识别颜色为白色
> 
> 用于导入：在 LibreCad 中默认为黑色背景上的白色笔 - 但您不想用白色笔绘图。为此，我可以选择使用白色作为黑色。
> 
> 如果要更改二维视图内的颜色，请查看此处：
> [image]

[image]
How does a black place become white

svenhb, 2021-07-28:
>How does a black place become white  

Only if you change color in original DXF file

lirui-dot, 2021-07-28:
> > 黑色的地方怎么变成白色的
> 
> 仅当您更改原始 DXF 文件中的颜色时

The original is white

svenhb, 2021-07-28:
Perhaps there is a bug, this option doesn't work...
[image]

[31 more replies]

closed 2021-09-04 by svenhb