svenhb/GRBL-Plotter #208: installation error

Karijn, 2021-07-25:
Installation problem

When i try to install the version 1.5.8.4 i get the following error

Fehler beim Herunterladen einer enfordlichen datei...

Fehler beim Herunterladen der folgenden Ressource:
http://go.microsoft.com/fwlink/?linkid=164199&clcid=0x411

Datum: 7/25/2021 1:44:52 PM

Weitere Informationen finden Sie in der Setup-Protokolldatei unter "C:\Users\Karij\AppData\Local\Temp\VSD801F.tmp\install.log".

**Screenshots**
[image]

svenhb, 2021-07-25:
Yes I heared about, but can't fix it at the moment. 
Workarround: the source code contains still the subfolder "GRBL-Plotter\bin\Release" which contains all files and the exe. Just start "GRBL-Plotter\bin\Release\GRBL-Plotter.exe" at least this should work

Karijn, 2021-07-25:
Hi Sven,

 

As a C# developer, I just compiled the source 😊

 

Ps. The installer is only in German, can you make English possible too?

 

 

Met vriendelijke groet/with kind regards/mit freundlichen Grüßen/Avec la salutation aimable/Saygilanmizla/Barati udvozlettel/Saludos/敬具/Veliki pozdrav,

 

 

 

Karijn Wessing

Knoopkruid 134

1112 PZ Diemen

06-28 139 139

 

Dit bericht en de eventueel daarbij behorende bijlagen zijn uitsluitend bestemd voor de persoon aan wie het is gericht en kan informatie bevatten die persoonlijk of vertrouwelijk is. Verdere verspreiding, openbaarmaking of vermenigvuldiging van dit bericht is zonder voorafgaande toestemming van IN2TECH niet toegestaan. Indien dit bericht niet aan u is gericht, verzoeken wij u ons onmiddellijk telefonisch of per e-mail op de hoogte te stellen en het originele bericht te vernietigen. Gebruik door anderen dan geadresseerde(n) is verboden. Dank voor uw medewerking. 

This message, including any enclosures, is intended solely for the person to whom this message is addressed to and could contain personal or confidential information. Any further distribution, disclosure or copying of this message is prohibited, unless by preceding authorisation of IN2TECH. If you are not the intended recipient of this message, you are requested to inform us immediately by phone or E-mail and destroy the original message. Any use by anyone else than the addressee or addressees is prohibited. Thank you for your cooperation.

svenhb, 2021-07-25:
Fine,
I'm just working on a new installer and do also code clean up, to get rid of all (or most) of the warnings...
I assume I need to switch my visual studio to english to get english error messages...

svenhb, 2021-07-31:
Please check setup of new pre-release: https://github.com/svenhb/GRBL-Plotter/releases

svenhb, 2021-09-04:
I think it's solved with new installer

closed 2021-09-04 by svenhb