svenhb/GRBL-Plotter #210: Source code recompile error

lianzaozi, 2021-07-31:
Source code recompile error

[image]

svenhb, 2021-07-31:
no picture uploaded.
Do you have all the needed packages, not sure if they will be installed automatically.
[image]

lianzaozi, 2021-08-01:
The file "Dimensions" was not found, resulting in an error in compiling the source code

[image]

svenhb, 2021-08-04:
Is it fixed?

svenhb, 2021-08-05:
No idea, sorry.

lianzaozi, 2021-08-05:
Thank you very much for your reply. I am looking into the reason myself.

svenhb, 2021-09-03:
Could you fix it? I assume mixing file versions in same folder (e.g. "xmlMarker.cs" = old version, "XmlMarker.cs" = new version)
There where to files with same name but the right one (new version) starts with capital letter.

svenhb, 2021-09-04:
Probably fixed

closed 2021-09-04 by svenhb