svenhb/GRBL-Plotter #211: Why do I dynamically change G code and he becomes what he was

lirui-dot, 2021-08-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

svenhb, 2021-08-02:
I don't understand this issue

lirui-dot, 2021-08-02:
> 我不明白这个问题

You change the G code in the code, and it becomes the original G code

lirui-dot, 2021-08-02:
> > 我不明白这个问题
> 
> 你把代码里的G代码改了，就变成原来的G代码了

There is also the box can choose graphics, or choose multiple graphics

svenhb, 2021-08-02:
I still don't understand, do you talk about transformations? Or import options?
Please give screenshots and examples.

lirui-dot, 2021-08-02:
> 我还是不明白，你说转换吗？还是导入选项？
> 请给出截图和例子。

Can I choose more than one graphic

lirui-dot, 2021-08-02:
> > 我还是不明白，你说转换吗？还是导入选项？
> > 请给出截图和例子。
> 
> Can I choose more than one graphic

Probably got into the timer and changed the G code

svenhb, 2021-08-02:
No: only one figure (marked yellow) or all figures = whole graphics

svenhb, 2021-08-02:
If you use grouping, I think all figures within a marked group can be trnasformed

lirui-dot, 2021-08-02:
> 如果使用分组，我认为标记组内的所有数字都可以转换

Like washing when the knife into the knife back where to set

lirui-dot, 2021-08-02:
> > 如果使用分组，我认为标记组内的所有数字都可以转换
> 
> Like washing when the knife into the knife back where to set

像洗的时候进刀退刀量在哪设置呢

lirui-dot, 2021-08-04:
How to set the amount of edge wash in and out

lirui-dot, 2021-08-04:
> 如何设置洗边进出量

我看你这边的进刀量是0.1

lirui-dot, 2021-08-04:
> > 如何设置洗边进出量
> 
> 我看你这篇的进刀量是0.1

How do you set it up

svenhb, 2021-08-04:
Please show a screenshot of the settings you are talking about. Just paste an image from the clipoard here...

lirui-dot, 2021-08-04:
> 请显示您正在谈论的设置的屏幕截图。只需在此处粘贴剪贴板中的图像...

I mean there is a compensation for the amount of knife in and out when washing the edge

lirui-dot, 2021-08-04:
> > 请显示您正在拍摄的影像设置的屏幕截图。
> 
> 我的意思是清洗边缘时进出刀的数量是有补偿的

I don't know if yours has this function

lirui-dot, 2021-08-04:
> > > 请显示您正在拍摄的影像设置的屏幕截图。
> > 
> > 
> > 我的英文是锁边时进出刀的数量是有证明的
> 
> 不知道你的有没有这个功能

我的意思是在洗边的时候是否有进刀退刀量这个参数

svenhb, 2021-08-04:
I think you mean offset from center?
https://github.com/svenhb/GRBL-Plotter/wiki/Drag-tool-compensation
[image]

lirui-dot, 2021-08-04:
> 我想你的意思是从中心偏移？
> https://github.com/svenhb/GRBL-Plotter/wiki/Drag-tool-compensation
> [image]

> 我你的英文是从中心偏移想？
> https://github.com/svenhb/GRBL-Plotter/wiki/Drag-tool-compensation
> [image]

I'll try

svenhb, 2021-08-04:
Yes the distance from center to tip of the tool (brush or knife)
[image]

lirui-dot, 2021-08-04:
[image]
这里设置吗

svenhb, 2021-08-04:
[image]

lirui-dot, 2021-08-04:
[image]

svenhb, 2021-08-04:
This looks correct: if the ramp z-value is below zero, the path will be drawn inside the 2-d view

lirui-dot, 2021-08-05:
Can I choose more than one graphic

svenhb, 2021-08-05:
No - sorry, all or one figure.
But I will try to implement.

lirui-dot, 2021-08-05:
> 不 - 对不起，全部或一个数字。
> 但我会努力实现。

This should be easy to implement

lirui-dot, 2021-08-16:
[image]
Can I set the XY center axis according to a graph

lirui-dot, 2021-08-16:
[image]
Let's say I want to set it to this place in my circle based on this square

svenhb, 2021-08-16:
You can change the offset in your GCode using this:
[image]
or this:
[image]

Or you can apply an offset to your coordinate system, if a grbl controller is connected: https://grbl-plotter.de/index.php?id=form-coordinates&setlang=en

lirui-dot, 2021-08-17:
> 您可以使用以下内容更改 GCode 中的偏移量 ：
> [image]
> 
> [image]
> 
> 或者，如果连接了 grbl 控制器，您可以对坐标系应用偏移量：[https](https://grbl-plotter.de/index.php?id=form-coordinates&setlang=en) ://grbl-plotter.de/index.php?id [=](https://grbl-plotter.de/index.php?id=form-coordinates&setlang=en) form-coordinates&setlang [=](https://grbl-plotter.de/index.php?id=form-coordinates&setlang=en) en[](https://grbl-plotter.de/index.php?id=form-coordinates&setlang=en)

Is it possible to move according to a particular pattern

lirui-dot, 2021-08-17:
[image]
What is the order of the click graph

lirui-dot, 2021-08-17:
> [image]
> What is the order of the click graph

Can you set it to a graph a grap

lirui-dot, 2021-08-17:
[image]
If I just want to dot one line here, it all changes color

lirui-dot, 2021-08-17:
[image]
[image]
Cad和导入的不一致

lirui-dot, 2021-08-17:
[image]
And what was the reason you said before

svenhb, 2021-08-17:
>What is the order of the click graph 

If you enable pen-up path, you will see numbers of figure id.
[image]

>If I just want to dot one line here, it all changes color 

Try selection modifier:
[image]

>And what was the reason you said before 

Division by zero, but was fixed: https://github.com/svenhb/GRBL-Plotter/issues/207
 Which version you are using? 
Try 1.6.0.2/3 https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2

lirui-dot, 2021-08-17:
> > 点击图的顺序是什么
> 
> 如果您启用画笔路径，您将看到人物 ID 的数字。
> [image]
> 
> > 如果我只想在这里画一条线，它都会改变颜色
> 
> 尝试选择修饰符：
> [image]
> 
> > 你之前说的原因是什么
> 
> 除以零，但已修复：#207
> 您使用的是哪个版本？
> 尝试 1.6.0.2/3 https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2

What do I need to change if I want to make changes in the previous version

lirui-dot, 2021-08-18:
[image]
The latest version of this DXF has an error
[6-4-282-24.txt](https://github.com/svenhb/GRBL-Plotter/files/7004153/6-4-282-24.txt)

lirui-dot, 2021-08-18:
What is the minimum computer that the latest version does not support

lirui-dot, 2021-08-18:
> 最新版本不支持的最低电脑是多少

There is also a problem is the complex graphics too card, card burst

svenhb, 2021-08-18:
>What is the minimum computer that the latest version does not support

Win XP is ok, but you can't use the installer. Instead unzip the ZIP file.

>There is also a problem is the complex graphics too card, card burst 

Can you explain?

lirui-dot, 2021-08-18:
[测试.txt](https://github.com/svenhb/GRBL-Plotter/files/7005184/default.txt)
Try using this file

lirui-dot, 2021-08-18:
> > 最新版本不支持的最低电脑是多少
> 
> Win XP 可以，但不能使用安装程序。而是解压缩 ZIP 文件。
> 
> > 还有一个问题就是复杂显卡太卡，卡爆了
> 
> 你可以解释吗？

You can do it in Xp but you have to generate a file

svenhb, 2021-08-18:
>Try using this file 

It works on my old (14 years?) I5 notebook.

>You can do it in Xp but you have to generate a file 

Idea: you have your CNC plotter / router in the basement and control it with an old notebook/pc (e.g. with Win XP) via GRBL-Plotter.

lirui-dot, 2021-08-18:
> > Try using this file
> 
> It works on my old (14 years?) I5 notebook.
> 
> > You can do it in Xp but you have to generate a file
> 
> Idea: you have your CNC plotter / router in the basement and control it with an old notebook/pc (e.g. with Win XP) via GRBL-Plotter.

Today's computers are also very slow

lirui-dot, 2021-08-18:
> > > Try using this file
> > 
> > 
> > It works on my old (14 years?) I5 notebook.
> > > You can do it in Xp but you have to generate a file
> > 
> > 
> > Idea: you have your CNC plotter / router in the basement and control it with an old notebook/pc (e.g. with Win XP) via GRBL-Plotter.
> 
> Today's computers are also very slow

I just sent you that file

lirui-dot, 2021-08-18:
> > > > 尝试使用此文件
> > > 
> > > 
> > > 它适用于我的旧（14 岁？）I5 笔记本。
> > > > 您可以在 Xp 中执行此操作，但必须生成一个文件
> > > 
> > > 
> > > 想法：你在地下室有你的 CNC 绘图仪/路由器，并通过 GRBL-Plotter 用旧笔记本/电脑（例如使用 Win XP）控制它。
> > 
> > 
> > 现在的电脑也很慢
> 
> 我刚把那个文件发给你

You can try it. Is it stuck

svenhb, 2021-08-18:
I updated both setups here: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2

lirui-dot, 2021-08-18:
> 我在这里更新了两个设置：[https](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2) : [//github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2)

That doesn't seem to help my problem

svenhb, 2021-08-18:
I think your file-sizes are very huge, compared to the visible content. Perhaps you can export your drawings to SVG and then test import performance of GRBL-Plotter?

lirui-dot, 2021-08-18:
> 我认为与可见内容相比，您的文件大小非常大。也许您可以将您的绘图导出为 SVG，然后测试 GRBL-Plotter 的导入性能？

But we mainly have more needs for DXF files

lirui-dot, 2021-08-18:
> > 我认为与观察到的内容非常大，您可以将您的绘图导出为 SVG，然后测试 GRBL-Plotter 的导入性能？
> 
> 但是我们主要是对DXF文件有更多的需求

How to Export SVG

svenhb, 2021-08-18:
>That doesn't seem to help my problem 

What exactly is the problem? This file 测试.txt ha a size of 2414 kB and took 5 sec. to be first visible and 12 sec. to be fully loaded - on my old I5.

svenhb, 2021-08-18:
>How to Export SVG 

I don't know your CAD software, perhaps it's not possible.

svenhb, 2021-08-18:
>I updated both setups here: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2 

This link is now for version 1.6.0.4 with fix for this 
[image]

lirui-dot, 2021-08-18:
> > 我在这里更新了两个设置：[https](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2) : [//github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2)
> 
> 此链接现在适用于 1.6.0.4 版，并对此进行了修复
> [image]

I'm using the latest version

lirui-dot, 2021-08-18:
> > > 我在这里更新了两个设置：[https](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2) : [//github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2](https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2)
> > 
> > 
> > 此链接现在适用于 1.6.0.4 版，并在此进行了修复
> > [image]
> 
> 我正在使用最新版本

Will the latest version support Windows 7 and XP

svenhb, 2021-08-18:
>Will the latest version support Windows 7 and XP

I think so

lirui-dot, 2021-08-18:
> > 最新版本是否支持 Windows 7 和 XP
> 
> 我认同

This version can support both

[25 more replies]

closed 2021-10-04 by svenhb