svenhb/GRBL-Plotter #212: Create G-Code from Text SVG/LFF Freezes 1.6.0

sjsawatzky, 2021-08-03:
**Describe the bug**
Attempting to create G-Code from Text using SVG or LFF fonts hangs/freezes/locks up the program.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy fonts folder from 1.5.8.4 data directory to 1.6.0.0 data directory
2. Open GRBL Plotter Ver.:1.6.0.0
3. Go to 'G-Code Creation' menu
4. Click on 'Create Text'
5. Enter Text
6. Select a SVG or LFF Font
7. Click 'Create G-Code' button

**Expected behavior**
I expect G-Code to be created from the text in the selected font

**Screenshots**
[image]

svenhb, 2021-08-03:
Thanks for finding this bug, I made a mistake with the folder structure.
If you copy the svg files to a subfolder svg it should work like ...\data\fonts\svg\svg\*.svg

svenhb, 2021-08-03:
If you open the header section you also should get information:
[image]

sjsawatzky, 2021-08-03:
If it was not frozen/hung, I could see what is in <Header > :-)

I have moved the svg folder to a svg subfolder; it now looks like:
\GRBL-Plotter\data\fonts\svg\svg\EMSAllure.svg

It still seems to be freezing/hanging. Is there any debugging log I can turn on to get you more information?

svenhb, 2021-08-03:
You will find a logfile inside the \GRBL-Plotter folder (where the subfolder "data" is located).
Hope it helps. When does the freezing occurs?

sjsawatzky, 2021-08-03:
It freezes after clicking the 'Create G-Code' button.

This is the last log entry:
 | Error | GrblPlotter.GCodeFromFont      |  Font 'svg\svg\EMSNixishItalic.svg' or file 'C:\Program Files (x86)\GRBL-Plotter\data\fonts\svg\svg\svg\EMSNixishItalic.svg' not found

This is a screenshot of the svg/svg folder, but it looks like another /svg has been added....
[image]

sjsawatzky, 2021-08-03:
As a workaround, I can create a copy of the SVG folder (instead of moving it) and then selecting the font in the parent SVG folder.

Thanks for your help and for your work developing such a useful program

Thanks

svenhb, 2021-08-03:
Thats wired... How did you install the program? Via Setup? I'm wonering about the data-path "C:\Program Files (x86)\GRBL-Plotter\data\f" usually this path is restricted and will not be choosen as a data path. But anyway this is probably not the main problem.

svenhb, 2021-09-04:
I think it's fixed

closed 2021-09-04 by svenhb