svenhb/GRBL-Plotter #213: writing an external program to communicate with grbl plotter

futurau, 2021-08-06:
Thanks for your great work again

I want to mount a camera at the center of and above the machine so that FOV will be entire XY plane. After placing a picture on table, camera will recognize the registration marks around picture and machine will immediately start cutting without further involvement of user.

I would like to create an external python program and communicate with GRBL Plotter.

For this, I must somehow two-way communication for data transfer and control GRBL Plotter. Do you have any leads?

svenhb, 2021-08-06:
>I must somehow two-way communication for data transfer and control GRBL Plotter.  
 
No idea for the moment. Perhaps this is a start point: https://docs.microsoft.com/en-us/windows/win32/ipc/interprocess-communications

svenhb, 2021-08-06:
How much data must be transfered?
For my extensions I use the clipboard: https://grbl-plotter.de/index.php?id=form-extenions&setlang=en

cem-dedetas, 2021-08-23:
Hi @svenhb , 
Thanks for the recommendation on inter-process communications. We wanted to be able to connect to remote networks/computers, the flexibility of bi-directional communication and the scalability, so we went with WebSockets solution on our fork.

closed 2021-10-04 by svenhb