svenhb/GRBL-Plotter #214: program stopped in middle of cutting,

engine3061, 2021-08-10:
program stopped and machine stopped in the middle of cutting, is there a way to start the cutting back at a certain line in the gcode?

svenhb, 2021-08-10:
Don't know if it works after break up, but after pause: you find a _processed.nc file in Menu-Open recent file.  
This file contains the GCode from editor when job was started.  
During a pasue the line number will be stored. 
If you load the _processed.nc file, you will be asked to continou at the last line. 
 
You may send me the logfile, to find the problem - located in the data folder (check the about window)

engine3061, 2021-08-10:
i ran it after to see if it would start from spot it stopped but it started from beginning, and went into about and didnt see any data folder, i just deleted the lines out it already cut in editor and restarted it from there and seems to be working fine, not sure if you can put a run from command it where we can just type in what line we want to start from in cases like this

engine3061, 2021-08-10:
also is there a delay i can put in for when the spindle goes to the starting point from the time i hit start, so the spindle can get u to speed before it starts without hitting the start spindle box

svenhb, 2021-08-10:
>not sure if you can put a run from command it where we can just type in what line we want to start from  

It's difficult, because the grbl-controller doesn't know the modal states of the code before. E.g. Line 1 starts with switching on the spindle, but you start the code at line 5 - expecting the spindle is on, but it's not... You would break your tool.

svenhb, 2021-08-10:
>also is there a delay i can put in for when the spindle 
 
If you import a graphic, you can set it:
[image]

svenhb, 2021-09-04:
I added the feature to start anywhere inside the code:
https://github.com/svenhb/GRBL-Plotter/releases

svenhb, 2021-11-01:
No feedback is a good feedback 😉

closed 2021-11-01 by svenhb