svenhb/GRBL-Plotter #215: dxf conversion causes circles to have weird shapes

yagizzha, 2021-08-11:
When using a dxf file to convert into GRBL plotter coordinate system circles have a weird shape , program used to create dxf file is "Illustrator CC 2018"

[example04.zip](https://github.com/svenhb/GRBL-Plotter/files/6969186/example04.zip)
[image]
[image]

svenhb, 2021-08-11:
I found your example right now, yes there is a problem with splines....
I use LibreCad and had never such a problem.

svenhb, 2021-08-12:
Very strange dxf file: 
- LibreCad can display it, but not really working with
- inkscape can't import it (I use this algorithm to import splines) 
 
I think I can't fix it - you should try to avoid splines in your Illustrator CC 2018

hamppi, 2021-08-15:
Hi, Sven! 
Also a dxf connected question. Does it support import of Z values or is it some problem with my file? Whenever I try with different ways dxf imports without Z values.
Example with simple poliline.
[testpanckake_dxf.txt](https://github.com/svenhb/GRBL-Plotter/files/6989021/testpanckake_dxf.txt)

svenhb, 2021-08-16:
No, I don't read Z information from DXF file.
In LibreCad I just see one line with our example...

amoineau, 2021-08-16:
Hello,
I wanted to jump on this issue instead of creating a new one because I encountered a problem that seems similar.
Someone sent me a dxf file that should look like this : 
[B+v3 27g inter.pdf](https://github.com/svenhb/GRBL-Plotter/files/6990811/B%2Bv3.27g.inter.pdf)

but in grbl-plotter it looks like this : 
[image]

As you can see the oval are not oval and they are not closed. Do you know what might cause this issue ? Do you think that changing the way the dxf was created could solve the problem ?

I tried tinkering with the aforementioned settings but it didn't improve anything.

Here is the dxf : 
[b_v3_27g_inter_5_12_1_25_4_13_9.zip](https://github.com/svenhb/GRBL-Plotter/files/6990893/b_v3_27g_inter_5_12_1_25_4_13_9.zip)

When I open it with LibreCAD the ovals are not really oval either but still much better than in GRBL-Plotter. When my colleague opens it with Creo it appears just like the pdf.

svenhb, 2021-08-16:
It is a problem with the import of splines. I took the algorythm from Inkscape, which also has a problem importing this dxf.
Need to look for another algorythm...
[image]

amoineau, 2021-08-16:
Ok thank you, do you know if there is a way to work around that when creating the DXF ? Like transforming the spline into segments ?

svenhb, 2021-08-16:
Perhaps you can export to SVG? Perhaps your CAD will do a conversion of the splines.

svenhb, 2021-08-16:
Perhaps you can save the DXF in an older DXF format?

hamppi, 2021-08-16:
> No, I don't read Z information from DXF file.
> In LibreCad I just see one line with our example...

Yes its just one polyline but with Z values so its looking like this for me
[image]
I managed to create straight gcode from the curves and this works sure with Z also, but then I loose all the grbl-plotter functionality unfortunately ( so I have to manually create some additional paths and its getting a bit too complicated, I liked using subrouting and all that auto things. Is there other options to get Z value on import? Tried csv but it seems to be very specific with it... or shall I try more on it, it supports Z values? Other ways to export from rhino is -
[image]

svenhb, 2021-08-16:
I tried to implement the readout of the Z value (at least for the polyline you used). Unfortunatly LibreCAD can't show the Z axis...
Which CAD software do you use?

Try https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2

[image]

amoineau, 2021-08-17:
Ok I'll try those, thank you.

svenhb, 2021-08-17:
Note: your example testpanckake_dxf.txt uses positive Z values, which will be takeover to GCode.

svenhb, 2021-08-17:
Bug fix: I updated the setup in the 1.6.0.2 release

hamppi, 2021-08-17:
> I tried to implement the readout of the Z value (at least for the polyline you used). Unfortunatly LibreCAD can't show the Z axis...
> Which CAD software do you use?
> 
> Try https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.0.2

Wow nice. Seems to be working just fine at first tries! Thanks. 
I use Rhinoceros 3D for dxf export.

> Note: your example testpanckake_dxf.txt uses positive Z values, which will be takeover to GCode.

What do you mean? Seems its fine for me, Its a piece of brush way curve, so it stays above paint surface most time.

closed 2021-09-30 by yagizzha