svenhb/GRBL-Plotter #217: Improve graphic visualization (specifically with tiling)

amoineau, 2021-08-24:
**Don't superpose tiles when using the tiling feature**

Currently, when using the tiling feature every tile is displayed on top of each others, which makes the graph completely unreadable. Would it be possible to have the graph to be displayed "as is"?

**Form color**

Would it be possible for the forms to be drawn using the layer / tool color? This would allow for a much better understanding of what is going to happen and immediately notice if a form is not matched with the correct tool.

amoineau, 2021-08-30:
I just wanted to specify that hiding the shapes that are not on the current tile could be a solution too. If it's simpler to achieve.

svenhb, 2021-08-30:
>I just wanted to specify that hiding the shapes that are not on the current tile could be a solution too. If it's simpler to achieve. 

I need a good reason / context / switch: why to hide figures in the 2D view that are programmed in G-code.
What is your command to move to next tile - maybe that's a starting point to seperate the tiles.

amoineau, 2021-08-31:
I see, the command I run between tiles is (^3 $CP1000). It tells the conveyor to move forward 1000 mm.

svenhb, 2021-09-04:
I added the suggested features:
https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2021-09-06:
Wow thank you, I'll try that today !

amoineau, 2021-09-06:
Once again, it's some really nice work :

* The 2D view figure color is great, thank you !
---
* The display of tiles is a real improvement... with some limits ^^'
* Always seeing the original shape, instead of it folding, is obviously the big improvement !
* When the shapes of the second+ tiles are being processed the fact that the pink cursor is on the "real shape" plan and the black one is on the "machine" plan is actually really cool because you keep the best of both world : "where is the head IRL, and what am I drawing (or at least about to draw) ?".
* The big problem I see is that the green drawing is happening on the machine plan. Which makes things as messy as before after the first tile and go against the idea above, it's quickly impossible to follow what as been done, what is about to be done and even where is the head. 

[image]

(Here I chose a very hollow file with only two tiles just to keep the readability, but imagine 6 very busy tiles ^^).

It would be amazing if the green tracing happened on the "real shape" plan. Do you think it would be remotely possible ? That way we would have every possible information with complete readability !

I'm sorry to ask so much of you, I would understand if you wanted to keep it there. I just feel like the tile feature would be so neat and complete with that. (I hope I am not the only one using it because it would be a shame)

svenhb, 2021-09-06:
>It would be amazing if the green tracing happened on the "real shape" plan. Do you think it would be remotely possible ? That way we would have every possible information with complete readability ! 

I understand the problem, but not your suggested solution. Perhaps: Within the working area you want only to see the figures of the actually proccessed tile?

amoineau, 2021-09-06:
I made the worst paint edit in hope that it would be clearer than words x) :

[image]

I wished the green tracing that overwrites the processed shapes, would actually overwrite the shapes (even after the first tile). Right now the green lines are all stacked on the "machine" plan. 

I don't know if this is any clearer, don't hesitate to ask again ! ^^

amoineau, 2021-09-06:
The drawback would be that the black cursor would be moving alone in the machine plan, but hey, you can't have it all ;)

svenhb, 2021-09-06:
Like this? https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20190906.exe

amoineau, 2021-09-07:
I got this :

[image]

(The exterior of all sails has been processed on both tiles)

The green tracing doesn't stack on the machine plan but it doesn't seem to appear elsewhere either :(
And in the areas I circled there are some artefacts, but maybe it is not related to the changes.

svenhb, 2021-09-07:
Can you try to hide the limit area? I think it's a problem with the red grid.

amoineau, 2021-09-07:
I got a similar result :

[image]

Is that what you wanted ?

svenhb, 2021-09-07:
How does it look, if you simulate the path?

svenhb, 2021-09-07:
This means:
- with simulation all figures are marked correctly one by one tile by tile? (don't look at the "glitches") 
- on real process it only marks the figures within the working space 

Is this correct?

amoineau, 2021-09-07:
Yes, I will try with a file more than two tiles long just to make sure.

svenhb, 2021-09-07:
Good to know, will try to fix it

amoineau, 2021-09-07:
Thank you ! Good luck ^^'

svenhb, 2021-09-07:
Try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20210907.exe

amoineau, 2021-09-08:
I'm running the tests but for now the thing I noticed is that the dark green lines representing the pen path are slightly broken :/
 
[image]

amoineau, 2021-09-08:
It worked on that file !

[image]

svenhb, 2021-09-08:
Little improovement with pen-up path: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20210908.exe

amoineau, 2021-09-10:
Hi, sorry I am on a business trip since yesterday, I'll check it next week. Thank you again :)

amoineau, 2021-09-16:
Hi ! I finally got time to test it and it seems to be working well, thank you :)

I showed the previous version to the people who are going to use the machine and they found it confusing that the graphic doesn't shift left into the machine limit when moving to the next tile...
I know we are straying further and further from the original purpose of your software but do you think it would be hard to offset the visual after each tile ?

What they want would be to always have the current tile inside the machine limit (and that way the black cursor would stay on the graphic). I'm really hoping they will be satisfied after that so you can close this issue :/

amoineau, 2021-09-28:
Hello, did you consider my request ? I would understand if it was a no ^^'

svenhb, 2021-09-28:
>What they want would be to always have the current tile inside the machine limit (and that way the black cursor would stay on the graphic).

I tried to implement, but you have to live with strange behavior of the progress path. You need to wait until the next release - perhaps next weekend.

amoineau, 2021-09-28:
Ok sure, It's really not a pressing matter ! I simply wasn't sure you saw my last messages considering I took quite some time to answer yours last time, you might have moved on ^^
Thank you.

svenhb, 2021-10-09:
Improoved behavior of progressed path display: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_6_2_1.exe

svenhb, 2021-11-01:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases
Perhaps a bit improovement of behavior if "Move the tiles" is enabled.
[image]

closed 2021-11-01 by svenhb

amoineau, 2021-11-02:
Ok, I'll try that as soon as I can, thank you.

amoineau, 2021-11-02:
I tried it and it seems to have two problems (which might be the same) :
* When I don't use "Moves the tiles to 0;0", the light green tracing representing what has been done stays on the first tile even when doing the second one. However I recall that at some point you succeeded in making sure that this tracing followed the current progression in this configuration.
* When using "Moves the tiles to 0;0", after shifting, the light green tracing disappear for the previous tile and the one for the next tile is drawn on the first tile instead of the "0;0 area" ( which looks even weirder since the tile is shifted left, in the negative positions) :

[image]

Here we can see the green light tracing for the first tile is missing and is currently being drawn on the current position of the first tile.

Apart from that, having the option to switch between both mode is really great and everything else seems to be working very well !

svenhb, 2021-11-02:
Hard to understand, perhaps you can make a short video?
With my test data it looks as expected.

amoineau, 2021-11-02:
Short video will be hard, the files are rather long to process and i'm really not well versed in video capture with what I have at work. But I will try to make step by step captures to explain better. In the meantime here is the file if you want to try it and see if you get a different result than my previous screenshots :

[dxf-sample-amoineau.zip](https://github.com/svenhb/GRBL-Plotter/files/7461197/dxf-sample-amoineau.zip)

Maybe  some settings are interfering with each other, in my config and not yours...

amoineau, 2021-11-02:
When I start, the first tile is processed normally :
[image]

The the first is finished, the green tracing is perfect :
[image]

Then the graph is shifted to the left, as expected, but the green tracing drawn until then disappears :
[image]

Now we are processing the second tile, and the green tracing is being drawn correctly, albeit one tile to the left : 
[image]

At the end of the second tile we have this : 
[image]

amoineau, 2021-11-02:
Does this help you understand ?

svenhb, 2021-11-02:
>Does this help you understand ? 

Yes 
>Then the graph is shifted to the left, as expected, but the green tracing drawn until then disappears :  

that is intentional, otherwise each tile needs to get a seperate "processed" path. 

>Now we are processing the second tile, and the green tracing is being drawn correctly, albeit one tile to the left : 

this looks wrong and is different to my intention and tests - the green path should appear in the work area of the current tile...

svenhb, 2021-11-02:
Perhaps you can also show your tool-table "laser_pen_default.csv"?

svenhb, 2021-11-02:
I edited the last release: https://github.com/svenhb/GRBL-Plotter/releases 
#### 1.6.2.3
- DXF Import add DXFText (before just DXFMText)
- Bug fix tile shift offset of processed path

amoineau, 2021-11-03:
> > Then the graph is shifted to the left, as expected, but the green tracing drawn until then disappears :
> 
> that is intentional, otherwise each tile needs to get a seperate "processed" path.

That's understandable, and really not a problem indeed, considering we already know on which tile we are when using this mode.

amoineau, 2021-11-03:
> Perhaps you can also show your tool-table "laser_pen_default.csv"?

Sure : 
[image]

amoineau, 2021-11-03:
> I edited the last release: https://github.com/svenhb/GRBL-Plotter/releases
>

Thank you ! I will try to test it today.

svenhb, 2021-11-03:
>Here is my config if it can help :  

BTW,, to reduce G-Code size and communication load,
I would use just 2 decimals
[image]

and enable "Remove very short moves" with perhaps 0.5? Should be accurate enough in this dimensions.
[image]

amoineau, 2021-11-15:
Thanks for the advice !

amoineau, 2021-11-15:
I unchecked the remove very short moves because it used to completely alter the shapes. I tried again and even with 0.05 it still does, here is a example (not the most visible one but is already too much).

This is the file without the option :

[image]

And this is the file with :

[image]

amoineau, 2021-11-15:
On a side note, the new version works really well ! I'll do more extensive test this afternoon and tomorrow but the first test is really satisfying.

svenhb, 2021-11-16:
I found a bug in "remove very short moves", will fix it soon.

amoineau, 2021-11-16:
Ok, cool !

svenhb, 2021-11-19:
I fixed the bug: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2021-11-22:
Ok cool I should be able to try it on Thursday !

commit fba54138 (2021-09-07): Version 1.6.1.1

Fix #217 and #220