svenhb/GRBL-Plotter #218: Does the current version have lathe machining

lirui-dot, 2021-08-30:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

svenhb, 2021-08-30:
No lathe machining, but rotary axis is supported

lirui-dot, 2021-08-30:
> 无车床加工，但支持旋转轴

How to set up

svenhb, 2021-08-30:
https://youtu.be/BLRfBk9yIbw

lirui-dot, 2021-08-31:
Why can't you debug it

svenhb, 2021-08-31:
Debug what?

lirui-dot, 2021-09-01:
> Debug what?

I just can't debug it

lirui-dot, 2021-09-01:
[image]

lirui-dot, 2021-09-01:
[image]
Can't debug

svenhb, 2021-09-01:
You use old version of CheckUpdate.cs

lirui-dot, 2021-09-01:
> 您使用旧版本的 CheckUpdate.cs

ok

lirui-dot, 2021-09-01:
Still can't

svenhb, 2021-09-01:
You may delete all your local  files and download again the project

lirui-dot, 2021-09-01:
> 您可以删除所有本地文件并重新下载项目

ok

lirui-dot, 2021-09-01:
Still can't

lirui-dot, 2021-09-01:
> 还是不能

debug Still can't

svenhb, 2021-09-01:
Did you try Release-mode?
[image]

lirui-dot, 2021-09-01:
> 您是否尝试过发布模式？
> [image]

NO

lirui-dot, 2021-09-01:
> > 您是否已经尝试过发布模式？
> > [image]
> 
> 不

You can't

lirui-dot, 2021-09-01:
> > > 您是否已经尝试过发布模式？
> > > [image]
> > 
> > 
> > 不
> 
> You can't

How to deal with this problem

svenhb, 2021-09-01:
Do you use VisualStudio 2019?
Did you also replace https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter.sln?

lirui-dot, 2021-09-01:
> 你使用 VisualStudio 2019 吗？
> 您是否也替换了https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter.sln？

yes

svenhb, 2021-09-01:
Current Version is with uppercase
[image]

lirui-dot, 2021-09-01:
> 当前版本为大写
> [image]

Also tried

closed 2021-10-04 by svenhb