svenhb/GRBL-Plotter #219: Tangential knife angle does not follow xy trajectory

lm150617, 2021-08-31:
The tangential knife angle does not follow the xy trajectory of my path. The blade stays at a 0 degree angle.

Changes to the default setup:

File/Setup/Graphic imports/Path imports/Path modifications
1. Add Tangential Axis (Enabled)
2. Name of tangential axis (Z)
3. Swivel angle (30)
4. Correct start angle if deviation is > (5)
5. Units per turn (3.6)

File/Setup/Graphic imports/G-Code generation
1. Z axis (Disabled)
2. PWM(Enabled)
3. Pen Up (50)
4. Pend Down (500)

I expected the blade to rotate as it passes through the rounded edge, like in this demo: https://youtu.be/uMFyTCK8Ggw

Screenshot:
[image]

svenhb, 2021-08-31:
This modification only works on graphics import. After import you see the applied modifications in the bottom line:
[image]
I don't see this in your screenshot, propably you loaded a GCode file, which will not be modified by this setup.

lm150617, 2021-08-31:
Im using Inkscape. I made a path to look similar to the one on your Youtube video. I imported as an .svg file.

[image]

svenhb, 2021-08-31:
Did you read my edited comment?
What do you see as import options in the bottom line after import?
[image]

lm150617, 2021-08-31:
I see :
[Import options: <Remove offset> <Tangent axis>]

lm150617, 2021-08-31:
I see:
[Import options:<Remove offset><Tangent axis>]

svenhb, 2021-08-31:
Can you upload your svg file?

lm150617, 2021-08-31:
I tried but github error says "We don't support that file type"  
Is a zip okay?
[test 1.zip](https://github.com/svenhb/GRBL-Plotter/files/7081821/test.1.zip)

svenhb, 2021-08-31:
Are you sure your settings here are same?
[image]

lm150617, 2021-08-31:
Yeah I checked it many times and tried to play aroud with the numbers and its the same every time.

lm150617, 2021-08-31:
[image]

lm150617, 2021-08-31:
The only way ive been able to make it work is doing a circle with a 90 
[image]
degree cw transformation.

lm150617, 2021-08-31:
The only way ive been able to make it work is doing a circle with a 90 degree cw transformation.

[image]

svenhb, 2021-08-31:
Your test1.svg works for me... What settings do have here:
[image]

lm150617, 2021-08-31:
[image]

svenhb, 2021-08-31:
I assume there is an additional modification activated, which is not listet in the bottom line...
There should be no difference with this issue, but who knows... Try latest Version 1.6.0.5: https://github.com/svenhb/GRBL-Plotter/releases

lm150617, 2021-08-31:
Thank you for taking the time to look at my issue. I appreciate you!

svenhb, 2021-09-03:
Any success up to now?

lm150617, 2021-09-04:
No, I've noticed that when I upload an svg the z axis stays at 0 unles its a circle. Also I don't see an up/down translation command at corners. Any idea why im not having success?

svenhb, 2022-04-01:
Perhaps it is solved, if your problem matches #270 or #272
[New features](https://github.com/svenhb/GRBL-Plotter/releases)

closed 2022-04-01 by svenhb