svenhb/GRBL-Plotter #220: Special conversion

trantai369, 2021-09-03:
Hi!
how to remove groove in curve
i want the grooves in the corner
can you suggest the code
thanks

[image]

svenhb, 2021-09-03:
Should work if you comment out lines 194, 195:
```
//                            else
//                            { distAccumulated += distance; }   // is curve, accumulate small distances

```

trantai369, 2021-09-03:
> Sẽ hoạt động nếu bạn nhận xét ra các dòng 194, 195:
> 
> ```
> //                            else
> //                            { distAccumulated += distance; }   // is curve, accumulate small distances
> ```

I made it
but not active
thanks you

trantai369, 2021-09-04:
I fixed many codes
but they don't work
someone help me

svenhb, 2021-09-04:
I  built in an option to switch off notches in curves. Will upload in some hours.

trantai369, 2021-09-04:
> I built in an option to switch off notches in curves. Will upload in some hours.

Thanks you !

svenhb, 2021-09-04:
https://github.com/svenhb/GRBL-Plotter/releases
[image]

trantai369, 2021-09-04:
Hi Svenhb !
doesn't seem to work well
[image]
[image]

svenhb, 2021-09-04:
It looks like, what I expect - no grooves / notches within the curves
[image]

And here with grooves / notches for easier bending:
[image]

trantai369, 2021-09-04:
I think it's more than one paragraph

[image]

svenhb, 2021-09-05:
For me it is correct:
I assume an endless band of material, so you can cut more letters in once.
1) the start cut will be made to have a defined edge
5) end of the object, cut off from material band 
[image]

trantai369, 2021-09-05:
> [image]

can you optimize it?
there is 1 more embryo to remove

[image]

Like the picture this picture is better

svenhb, 2021-09-05:
Startpoint depends on your graphics and import settings:
Switch off sorting to keep original start point from your drawing
[image]

trantai369, 2021-09-06:
Hi Svenhb !

[image]

svenhb, 2021-09-07:
You are wrong, explanation of "Min distance....":
Drawing with segment length of ca. 5:
[image]
If "Min distance..." is 4, every edge causes a notch:
[image]
If "Min distance..." is 6, segments will be summed up to create a curve - if enabled (it's not here):
[image]
With curves enabled:
[image]

Also Tool angle has an influence to the total length:
[image]

trantai369, 2021-09-07:
how to have numbers automatically ?  For some vectors it hasn't eliminated the "curve" completely.

svenhb, 2021-09-07:
>how to have numbers automatically ?  

https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20190906.exe

>For some vectors it hasn't eliminated the "curve" completely. 

Then you need to adapt the "Min distance" value

trantai369, 2021-09-07:
[image]

You can completely remove the curved points automatically. If tick "No curve".

svenhb, 2021-09-07:
I choose 10% of figure height: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20210907.exe

trantai369, 2021-09-07:
> I choose 10% of figure height: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20210907.exe

Thanks you
I checked, it's working fine now

trantai369, 2021-09-08:
[image]

svenhb, 2021-09-08:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1611_20210908.exe
[image]

trantai369, 2021-09-08:
You open source code ?
Thanks

svenhb, 2021-09-09:
I will collect improovements - then I make a new release with updated source code

trantai369, 2021-09-11:
[image]
[image]

Error detected
Same file, when increasing or decreasing "Min distance" length will be lost

svenhb, 2021-09-12:
A tool angle of 1° is not recommended I would say.... 
Better try with something usefull 30° to 120° which you could buy as a tool.

trantai369, 2021-09-12:
> A tool angle of 1° is not recommended I would say....
> Better try with something usefull 30° to 120° which you could buy as a tool.

I've done it but it doesn't work well
It works fine when setting "Min. distance" >=1 but this doesn't change the distance of edges
The "Z-Height" parameters do not work well, they run too many times at small distances, which is unnecessary. Just run 1 position according to the number of times Z is turned on.
You should see how the gcode length will change when adjusting these parameters.
The parameter "Tool angle" is not really necessary, just set it fixed, the paths will overlap, but no problem
The No cuver function works fine at this point.

trantai369, 2021-09-13:
Loss of length
I think the "Tool angle" should be fixed
The grooves are evenly divided according to the limit when "No curve" is removed.

[image]
[image]

svenhb, 2021-09-13:
The distance from notch to notch depends on tool angle and "Z-height for notch engraving": 
 - the larger the angle, the greater the distance 
 - the deeper the notch, the greater the distance
[image]

svenhb, 2021-09-13:
Can you upload your 33.svg file?
[image]

trantai369, 2021-09-13:
[image]

[33.zip](https://github.com/svenhb/GRBL-Plotter/files/7156500/33.zip)

svenhb, 2021-11-01:
Now with more information in the background path: https://grbl-plotter.de/index.php?id=form-setup#pathdevelopment

closed 2021-11-01 by svenhb