svenhb/GRBL-Plotter #221: Buggy Tool table

metzger100, 2021-09-07:
Hi Sven, 
thanks for your work. I really love the software and its possibilities, but I am facing a bug. 
All files are attached. I want to import the Metzger100electronics.svg with 50x50mm size into GRBL-Plotter to generate a GCode. 
[Metzger100electronics.svg.zip](https://github.com/svenhb/GRBL-Plotter/files/7122566/Metzger100electronics.svg.zip)
I bought a kit of drills for my cnc-router so I wanted to use them for my objects. I inserted all tools into the tool table and set the XYZ-Feedrates. In the Settings I enabled the Tool table and the grouping of the tooltable. I set the group properties to pen width and the sorting to group properties in inverted order so it goes from big to small. When I import the svg it says that the tool table is enabled but it only selects the 3.0 mm drill for the whole project. in the GCode only one group appears which is called like that:
`Group Id="1" PenWidth="" ToolNr="1" ToolName="3.0" PathLength="1752.0" PathArea="2500.1" `
[CNC.csv](https://github.com/svenhb/GRBL-Plotter/files/7122551/CNC.csv)
so it looks like the pen width is not set correctly but when I check the tool table there is content in the collumn. I tried to use 3,0 as well as 3.0 but it seems that it doesn't make any difference. The PenWidth stays empty in the GCode-Group. When I set a standard tool in the tool table settings the only thing that changes is the toolnumber. But the name or the penwidth doesn't change. For me it looks like a bug. Thanks for your help.
Metzger

svenhb, 2021-09-07:
Thanks for the detailed description. 
1) problem: checked via Inkscape: after ungrouping, set fill to white and set stroke to black I get the outlines - and all of them have the same thickness = pen width. 
2) problem: the svg file doesn't contain any stroke-width information - your pen width was not exported.

You should fix this first, before we can go on...
[image]

No stroke-width information:
[image]

Expected something like this:
[image]

metzger100, 2021-09-07:
Hi Sven,
I see. Thanks for the Infos. I set the stroke-width to 2.26771662 (0.8mm) now and ungrouped everything. And disabled the fill after I don't want the fill machined. 
[image]
Stroke appears here now
[image]
The Gcode has now one tool change but still to the 3.0mm drill and not to the 0.8mm drill. 
`<Group Id="1" PenWidth="0.8" ToolNr="1" ToolName="3.0" PathLength="1727.7" PathArea="2431.2>`
[image]
[Metzger100electronics2.svg.zip](https://github.com/svenhb/GRBL-Plotter/files/7123596/Metzger100electronics2.svg.zip)
Here is the updated svg.

svenhb, 2021-09-07:
All stroke-widths are same which results to one group with PenWidth=0.8 - as shown in your screenshot.
We expect as much groups as different pen-widths (or colors, or layers - depending on selected grouping)

metzger100, 2021-09-07:
Hi Sven, aaaaahhhhhh... Now I understand how the tooltable works.
Thanks for the explanation.

closed 2021-09-07 by metzger100

svenhb, 2021-09-07:
So finally you got the expected results? 👍👍👍

metzger100, 2021-09-08:
Hi Sven, now I understand how it works and have to try some stuff before I can tell if it works like i want it to work. If not I will open a new issue after this one is no issue and new one will be another topic. Thanks for your help.