svenhb/GRBL-Plotter #222: Center red circle function moves counter clockwise around the red circle

yagizzha, 2021-09-08:
Tried the center red circle function , it was just moving counter clockwise around it rather then actually centering it

svenhb, 2021-09-08:
I don't know which function you try to use.
Please give screenshots, the file you try to import, everything which helps to understand and follow the problem.

yagizzha, 2021-09-08:
[image]

This is the button/function i was trying to use , because of a electric outage i cant send the full screenshot

svenhb, 2021-09-08:
I didn't tried for a long time.... Perhaps you need to select a filter set first (upper right drop-down)? As seen on the screenshot here: https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling

closed 2021-09-10 by yagizzha