svenhb/GRBL-Plotter #223: Help with fiducial marking

yagizzha, 2021-09-10:
I am trying to automate the fiducial selecting process, 
wanted to ask at which part is the svg files drawings go into GRBL-Plotter's coordinate system and how i could mark them as fiducials

trantai369, 2021-09-10:
[image]
Is this the problem?
Please provide more photos

futurau, 2021-09-10:
We want to select a circle named with id="fiducial
For example: 
<svg xmlns="http://www.w3.org/2000/svg" width="210mm" height="297mm" viewBox="0 0 595.28 841.89">
  <defs>
    <style>
      .cls-1 {
        fill: #231f20;
      }
    </style>
  </defs>
  <title>test</title>
  <g id="fiducial">
    <circle class="cls-1" cx="118.5" cy="269.5" r="16.5"/>
  </g>
</svg>
How can we select the circle with a specific id after importing to GRBL-Plotter.

yagizzha, 2021-09-10:
[image]
when you use this function to select a circle for example , how does the code know which circle in the svg file is selected. 
Trying to have a system that puts dots to similiar places so i can automate it , but not sure on how to select them to start formatting

svenhb, 2021-09-11:
This dialog is for image recognition, check: https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling
[image]

svenhb, 2021-09-11:
>wanted to ask at which part is the svg files drawings go into GRBL-Plotter's coordinate system 

Check here: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/GCodeCreation/ImportFromFile

futurau, 2021-09-11:
can  we add (comments) in G-code?

svenhb, 2021-09-11:
No, but you may see SVG-id in group or figure XML attribute

svenhb, 2021-09-11:
[image]

yagizzha, 2021-09-13:
Another problem i have right now is not being able to mark circles as fiducials or not being sure if i picked or not  : 

[image]

On https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling it says i need to click on it to mark as teach point . 

[image]

on here the picked one has a pink cross on it

svenhb, 2021-09-13:
In my example the figures are drills (just a point in DXF file). The marker will only snap on coordinates which are given in the GCode.

svenhb, 2021-09-14:
In the next release the center of real circles (G2 / G3 GCode) can be selected.

closed 2021-09-16 by yagizzha